RavenSystem/esp-homekit-devices #90: ОТА

Kristian8606, 2018-11-19:
RavenSystem how can I upload my own BIN file through OTA

HomeACcessoryKid, 2018-11-21:
Can you indicated the file that is trying to be collected in your log

see issue #52 which might be the same thing

rechin304, 2018-11-26:
this is what i get, in a endless loop

erasing@0xf3000
00 00 f0 3f ... 00 00 00 14    
so far collected 418672 bytes
--- ota_verify_hash
--- ota_hash
--- ota_hash
--- looping
--- ota_set_verify...OFF
--- ota_get_version
--- ota_connect
target IP is 140.82.118.3 bind socket 58779....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HomeACcessoryKid/ota@version:"0.1.0"
--- ota_boot...0
--- ota_compare
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/ota/releases/download/0.1.0/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect
target IP is 140.82.118.3 bind socket 58780....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HTTP/1.1 302 Found
Date: Mon, 26 Nov 2018 19:43:53 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Server: GitHub.com
Status: 302 Found
Cache-Control: no-cache
Vary: X-PJAX
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/118357494/ce388712-40da-11e8-9077-695b312c7101?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20181126%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20181126T194353Z&X-Amz-Expires=300&X-Amz-Signature=71fac0143c6c75a5bb97f0f3511b73d8afa5da1a90285a155783f816d2000644&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dcerts.sector.sig&response-content-type=application%2Foctet-stream
Set-Cookie: has_recent_activity=1; path=/; expires=Mon, 26 Nov 2018 20:43:53 -0000
Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Fri, 26 Nov 2038 19:43:53 -0000; secure; HttpOnly
Set-Cookie: _gh_sess=cXo5QW1UN0JBWWFxcmdHaEZqNVBBMzliVWgwNlBJaTNaMFBPMnhPQ2N0KzI1dm9DTUZ1YzNQblZpdHdERW
HTTP returns 302
--- ota_connect
target IP is 52.216.238.51 bind socket 58781....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github-production-release-asset-2e65be.s3.amazonaws.com port 443....OK
send request......OK

8c 1d 63 ce ... 51 83 17 d7    
so far collected 155 bytes
--- ota_verify_hash
--- ota_hash
--- ota_hash
--- ota_set_verify...ON
certs size: 2628
TIME: Mon Nov 26 19:43:57 2018
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/ota/releases/download/0.1.0/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect
target IP is 140.82.118.3 bind socket 58782....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HTTP/1.1 302 Found
Date: Mon, 26 Nov 2018 19:44:00 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Server: GitHub.com
Status: 302 Found
Cache-Control: no-cache
Vary: X-PJAX
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/118357494/ce388712-40da-11e8-9077-695b312c7101?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20181126%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20181126T194400Z&X-Amz-Expires=300&X-Amz-Signature=54bdebe7e392f4b56567ef972a38510fd36a2967e1f4a6b2c4ec261e4f0c7e7c&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dcerts.sector.sig&response-content-type=application%2Foctet-stream
[... 594 lines cut ...]
send request......OK

--- ota_connect
target IP is 52.216.97.227 bind socket 58790....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github-production-release-asset-2e65be.s3.amazonaws.com port 443....OK

so far collected 409600 bytes
send request......OK

erasing@0xf1000
21 2b 92 a0 ... 21 1a 1c 83    4b 42 42 61 ... 27 93 5d 32    05 01 2c a2 ... 6c 37 d2 6c    38 2d 0c 45 ... 00 00 00 00    
so far collected 413696 bytes
send request......OK

erasing@0xf2000
00 00 00 00 ... 80 ba 2d 40    00 00 00 00 ... 00 00 00 00    00 00 00 00 ... 73 3a 00 00    61 30 20 25 ... 00 00 00 00    
so far collected 417792 bytes
send request......OK

erasing@0xf3000
00 00 f0 3f ... 00 00 00 14    
so far collected 418672 bytes
--- ota_verify_hash
--- ota_hash
--- ota_hash
--- looping

Kristian8606, 2018-11-26:
compile your BIN file (main.bin) and then create this file (main.bin.sig) with this script 'openssl sha384 -binary -out firmware / main.bin.sig firmware / main.bin
printf "% 08x" `as firmware / main.bin | wc -c` | xxd -r -p >> firmware / main.bin.sig 'then upload them to your repository, set the path to your repository, and you need to get it.

rechin304, 2018-11-26:
this log was from ota update

rechin304, 2018-11-27:
Some update: 
1. if i setup the OTA repository via a mobile phone (iphone) the OTA is looping
2. if i do the setup via iMac works fine (tested with 2 devices)

rechin304, 2018-11-27:
Thanks Raven i will keep a eye on it 👍

rechin304, 2018-11-27:
My findings are as mention early, if I do the setup of OTA repository by iMac (re flash 6 devices today and where all working) works ok. If I do the same setup of OTA repository via iPhone they enter in a loop.
A step by step instruction will be needed and i will gladly try to reproduce.

HomeACcessoryKid, 2018-11-28:
Instructions to extract the otamain sectors:
we know the file is 418672 bytes and that it starts at address 0x8d000
I will assume you know how to use terminal on a mac so you are in a terminal window in a directory where you can write files.
```esptool.py -p /dev/cu.usbserial-* read_flash 0x8d000 418672  otamain-in-flash.bin```
now download this file from ```https://github.com/HomeACcessoryKid/ota/releases/download/0.1.0/otamain.bin```
either take a checksum and compare. If equal, the files are equal
```
openssl sha384 otamain.bin
openssl sha384 otamain-in-flash.bin
```
if not equal, to know the details of where they are different, use xxd
```
xxd otamain.bin > otamain.bin.hex
xxd otamain-in-flash.bin > otamain-in-flash.bin.hex
diff otamain.bin.hex otamain-in-flash.bin.hex > otamain.diff
cat otamain.diff
```
If the differences in otamain.diff are small, quote it here, else just say it is huge

rechin304, 2018-11-29:
I try refreshing again. This time did went smooth
Checksum are the same

SHA384(otamain.bin)= b76bf3323189028c178c8f65f3b4f4b330b9bbd382b925afb6e235886f5b799e4d6baafde524b94b94b9dfd8d4c849cc
SHA384(otamain-in-flash.bin)= b76bf3323189028c178c8f65f3b4f4b330b9bbd382b925afb6e235886f5b799e4d6baafde524b94b94b9dfd8d4c849cc

I will try another device soon.

rechin304, 2018-11-29:
another 6 sonoff's SV latter and just work fine ... hmm

rechin304, 2018-11-29:
another sonoff basic and error

connected with Security, channel 1
dhcp client start...
ip:192.168.1.88,mask:255.255.255.0,gw:192.168.1.254
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
wifiready-done
--- ota_boot...1
OTAMAIN VERSION: 0.1.0, compiled Apr 15 2018 18:17:56
--- ota_init
Fatal exception (0): 
epc1=0x4029aad0
epc2=0x00000000
epc3=0x400043df
excvaddr=0x402c2c17
depc=0x00000000
excsave1=0x40101190
Registers:
a0 40101190 a1 3fff8570 a2  3fff85e0 a3  00000001
a4  4029aad0 a5  fbf8ffff a6  04000002 a7  3feffe00
a8  00000100 a9  0000001a a10 00000018 a11 04000102
a12 00000001 a13 3fff88c0 SAR 00000018

Stack: SP=0x3fff8570
0x3fff8570: 3fff85e0 a5a5a5a5 a5a5a5a5 a5a5a5a5
0x3fff8580: a5a5a5a5 a5a5a5a5 40100360 4010039e
0x3fff8590: 4010039e 4029aad0 00000033 40101190
0x3fff85a0: 3fff85e0 3fff8650 00000001 4029aad0
0x3fff85b0: fbf8ffff 04000002 3feffe00 00000100
0x3fff85c0: 0000001a 00000018 04000102 00000001
0x3fff85d0: 3fff88c0 00000000 3fff0658 00000018
0x3fff85e0: 3fff8650 00000000 3fff0658 4028fd00

Free Heap: 27068
_heap_start 0x3fff18e0 brk 0x3fff98c4 supervisor sp 0x40000000 sp-brk 26428 bytes
arena (total_size) 32740 fordblks (free_size) 640 uordblocks (used_size) 32100

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Writing default boot config.
Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done

compare check sum look identical
openssl sha384 otamain.bin
SHA384(otamain.bin)= b76bf3323189028c178c8f65f3b4f4b330b9bbd382b925afb6e235886f5b799e4d6baafde524b94b94b9dfd8d4c849cc
openssl sha384 otamain-in-flash.bin
SHA384(otamain-in-flash.bin)= b76bf3323189028c178c8f65f3b4f4b330b9bbd382b925afb6e235886f5b799e4d6baafde524b94b94b9dfd8d4c849cc

HomeACcessoryKid, 2018-11-29:
OK, not dismissing the subject, but some stuff we buy in china is not the best quality. I have a pile of broken esp12e in the graveyard which had a very short life. Please consider this in the analysis. 
So with that, do you think they are flukes or there is a deeper story?

BR,
HacK

rechin304, 2018-11-29:
You could be right. I still have another to test. Let's see. Thanks so far for you support.

closed 2019-01-27 by RavenSystem