RavenSystem/esp-homekit-devices #91: No Response after 5 - 6 commutations

pastaga84, 2018-11-19:
Hi,

I have issues with the lastest firmware (0.4.9 or 0.4.8) on my sonoff S20 outlet.
S20 crash after 5 - 6 commutations and i have no response in homeapp or Eve.

Someone have same problem ?

Sonoff S20 : S20 EU V1.3    2017-2-16

Thanks for your help

Kristian8606, 2018-11-22:
I think the device has enough capacitors from the manufacturer so the problem is most likely to be solved by the code

RavenSystem, 2018-11-26:
I use several S20 EU without any problem. But some time ago, a working S20 became no response after 5 - 6 commutations; issue was a hardware wifi problem and solution was replace device.

pastaga84, 2018-11-27:
Hi José,

I maked some tests with tasmota firmware and MQTT server on raspberry pi 3 (mqtt + homebridge), it's work nicelly. (No wifi loss after 5 - 6 commutations , work since 2 days)

Maybe this information helpful for a futur code optimization ... 

Thanks you.

RavenSystem, 2018-11-27:
With Tasmota, esp8266 only manages MQTT commands, but HomeKit commands require a lot of more resources due to high level encryption (MQTT lacks encryption). A faulty unit can work with light wifi usage, but not with heavy usage; at least, with my broken S20, I can connect to wifi and use it with simple UDP commands, but not HomeKit, and same HomeKit firmware works fine in my others S20. Why? I don't know.

In other hand, remove your device from Home App, and try to erase and flash your unit again with the last firmware following Wiki instructions.

pastaga84, 2018-11-27:
Hi José,

I uploaded firmware again on my S20. (v0.5.1)
It's seem works, many quickly commutations, no WIFI crash.

But if i loss or disable WIFI on my iphone and i reconnect this, sonoff s20 say "no response" during 2-3min before works again...

thanks for your support.

Open-Elec-RU, 2018-11-29:
Flashed two new S20s to 0.5.1 set them to Device Type 3 in the Eve app...   and both disappeared from the Home app next morning.
The button is behaving as expected - three second press to tun on and off.  Unplugged, waited 5 minutes, plugged in again - nothing (and no OTA mode device WiFi either).
Any ideas other than refreshing?  Leave them as Type 1 next time (I don't need the button functionality)?

PS At the same time I updated two basics to 0.5.1 and they are working just fine (one of them took two attempts and had to turn power off and on for one, but that's happened before).

Open-Elec-RU, 2018-11-29:
PPS When plugging in the Button LED goes green for 1 second. in case this matters.

RavenSystem, 2018-11-29:
@Moskvitch there is no way that a HomeKit device (certified or no certified ) is deleted from Home App by itself. Even if your perform a factory reset, accessory will keep in Home App as "no response". I don't know why your two S20 disappeared from Home App.

I suggest that you verify if S20 is connected to wifi, and try to add to HomeApp again. if not, you can perform a factory reset hold down button for 10 seconds until led turn on.

OTA update fails sometimes, and several attempts are need to perform the update. Sometimes, 5 or 6 attempts.

Moskvitch, 2018-11-29:
### The Aqara Chinese devil confirmed
It's been what?  Two hours?  One of my freshly re-added S20s is gone already.  The Aqara wireless switch setup to control both S20s lost all the key assignments, one S20 is still there in the Home app and controllable, and the other one is vanished completely from Homekit.
It's not showing "No response", it hasn't magically moved to another room, it simply **completely disappeared** altogether - there's no icon anywhere in the Home app!
The other one is still there, but I ain't gonna risk it, so I logged out of the Aqara online account.  Interestingly, another switch controlling a Sonoff Basic also lost key action assignments but the Basic is still there in the Home app.
I understand this is completely off-topic, but, yes, "there **is** a way that a HomeKit device (certified or no certified ) is deleted from Home App by itself".
Scary thing is this Aqara hub has the Homekit logo plastered all over its packaging.  So much for Apple's "slow but very rigorous" certification program :(

RavenSystem, 2018-11-29:
It seems that Aqara removes pair token from Home App, but not from device side. So, Sonoff is paired with a deleted token, and of course doesnt work with HomeKit until you factory reset it and a new pair detup is launched.

Moskvitch, 2018-11-30:
Not sure I follow, I am not that technical. Why would it want to delete the token?  And what would be the correct procedure?
But in any case, it has something to do with their buggy online servers.
Just like with deleted scenes before, my pronlems stopped once I logged out yesterday after re-adding the switch for the second time.  Re-created automations, reassigned key actions, all ok. Except I just found out the stupid thing also killed my homebridge WeMo switch :)
Will spend some time trying to figure out which “homebridge button” to press for 10 seconds 😂
But, all in all, the problem has clearly nothing to do with your great firmware!
Muchas gracias!
PS here are my troubles detailed in the Homekit subreddit, in case you might be interested (a new direction, maybe, I hope? 😉 after all, they are also a Chinese maker of cheap but quality hardware and just like Itead their software just sucks! - we need you! )
https://www.reddit.com/r/HomeKit/comments/a1myl4/more_scary_observations_of_the_updated_aqara/?st=JP3VK2S2&sh=9df68301