RavenSystem/esp-homekit-devices #93: D1 Mini as a Temp/Hum Sensor with no Switch

hmbarbosa, 2018-11-20:
Hello José 

Wondering if you could add an option to remove the switch from (6. Switch Basic + TH Sensor) I'm using this setup with a D1 Mini + DHT22 and works perfectly with a sensor connected on GPIO14. Is any change to hide the switch option as the main intention is to use the D1 Mini + DHT22 as a sensor only.

Thank you very much for your time expended on this project mate!!! that's Awesome.

rechin304, 2018-11-29:
Maybe additional repo for this :) as there is no such implementation directly, they all use homebrige as middle man. Temp, Hum, PIR are few example where D1 mini can be used.

RavenSystem, 2018-12-02:
@hmbarbosa Added to [TODO List](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List)

closed 2018-12-02 by RavenSystem

RavenSystem, 2018-12-02:
@hmbarbosa Sorry, but for now it's impossible to hide switch, because is the main service, and custom setup characteristic needs it; changing this is too complex because involves a lot of code. Removing from TODO list.