RavenSystem/esp-homekit-devices #95: Shelly 1 doesn't boot (connect to WiFi) on 230 V

mgoeller, 2018-11-20:
I can flash the firmware to a Shelly 1 successfully as described in the Wiki, even configure the device as type "Shelly 1" with the virtual accessory in the Eve App. All this is done while powering the Shelly 1 through the FTDI232 usb serial adapter. I can successfully ping the device as well. Of course I don't hear the relay clicking because the Shelly isn't powered by 230 V yet but I can see the relay toggling in the serial monitor.

When I now unplug the serial adapter and power the shelly with 230 V it doesn't even connect to my WiFi, e.g. I can't ping it. If I connect the Shelly 1 directly to 230 V after flashing the firmware, I can see the OTA-XXXXXX WiFi Access Point and connect / ping it. After entering the WiFi credentials and OTA repository the Shelly is supposed to reboot and connect to the WiFi, which it never does. When plugging it after this again to the serial adapter, everything works as intended.

Any suggestions how to debug? Or what I am doing wrong?

RavenSystem, 2018-11-20:
Actually, I haven’t a Shelly 1. I burned it, and I’m waiting for a new one. I will investigate this bug, because I’m very interested in a working HomeKit with those devices.

Codi878, 2018-11-21:
Hi Jose,
I confirm that bug, I also tested 4.9 firmware on shelly1 v2 and unfortunately, after uploading without OTA the module is not visible on 230V power supply. It is not visible as AP mode.

RavenSystem, 2018-11-22:
Info about Shelly 1: 
http://www.thesmarthomehookup.com/the-shelly-1-smart-relay-is-it-better-than-the-sonoff-basic/

Codi878, 2018-11-22:
HI Jose

Thank you for a nice article. Today I have tested again and I am really able to pair with homekit shelly1 when connecting low voltage to programming pins and I can control it. When connected to 230v it does not show signs of life in the home app. Where is a problem ?

RavenSystem, 2018-11-22:
IMHO The problem is the high voltage used by Shelly GPIOs. I'm going to release a separated firmware for Shelly 1, and drop it from RavenCore support.

RavenSystem, 2018-11-26:
RavenCore 0.5.0 has been released just now. Next step will be work on KellyTron firmware (Shelly 1 firmware with native HomeKit).

AramVartanyan, 2018-12-09:
I want to try out the Shelly 1 with a basic switch firmware, but I do not want to burn it out (because of the mains reference voltage for the GPIOs).
Jose, would you give me an advice how to configure the I/Os or if I can use that part of your unfinished KellyTron?

AramVartanyan, 2018-12-11:
How did you burned your Shelly1?

AramVartanyan, 2018-12-11:
Wow! You are lucky that you are not injured and the Mac is up and running!

Very good! Shelly looks like a good product and it will be nice to add it to the HomeKit family of devices.

Thank you for your great work!

RavenSystem, 2018-12-11:
Yes, my iMac is alive, but my FTDI adapter burned out too. Amazon sent another unit soon. I'm interested in Shelly 1, it's smaller than Sonoff Basic, and it has a GPIO dedicated to connect a mount-wall switch with 230VAC. I want to use these devices in all lights of my house.

Wissoation, 2018-12-16:
Hey Ravensystem which FTDI Adapter have you bought from Amazon for the IMac

RavenSystem, 2018-12-16:
@Wissoation this one: https://www.amazon.es/dp/B01N9RZK6I/

Codi878, 2018-12-19:
Shelly will be the best homekit upgrade in Home app

Codi878, 2018-12-27:
Hi Jose,

I uploaded the software and shelly works ok, but after some time it's no longer visible in the home app. It helps to disconnect from the power and reconnect again.

RavenSystem, 2018-12-27:
@Codi878 Mine works well. Now it's up for 27 hours without any issue.

Codi878, 2018-12-27:
Hmm, I do not know, but all my Sonoff work perfectly but Shelly1 ver2 on KellyTron firmware does not respond after a few minutes

RavenSystem, 2018-12-27:
Is connected to 230VAC or FTDI?

Codi878, 2018-12-27:
230v

RavenSystem, 2018-12-27:
Disconnect it from 230VAC and connect to FTDI to recollect logs.

Codi878, 2018-12-27:
Jose on 3,3V everything works good

RavenSystem, 2018-12-27:
Ok, try version 0.0.2 and comment.

RavenSystem, 2019-01-07:
KellyTron firmware has been discontinued. Now RavenCore supports Shelly1.

closed 2019-01-07 by RavenSystem