RavenSystem/esp-homekit-devices #98: Sonoff 4CH R2 and Water Valve

GPL71, 2018-11-25:
Hello ....
I'm without words each time you release a new firmware .....
This is simply a great work.
I'm working adding sonoff everywhere in the house .... 😄 

Now I'm working on Water Valve in the garden.
I've 4 different valves. 
Now ...after I setted water valve only one channel (CH1) is available .
If I change in 4ch then I have 4 switches and manage each one of them.
Then ...
How to manage 4 water valve? I would like use timer for all devices.
Where did I wrong?

Thank you

Partizan7676, 2018-11-26:
make automation in Home App. Add them to one scenario. In eve app - you get more conditions to start scenario.

RavenSystem, 2018-11-26:
@GPL71 You did not do anything wrong. Device type 4 is for 4 single switches, but Sonoff 4ch in valve mode will never happen. Manage 4 individual timers with an esp8266 and HomeKit running is not a good idea due to low resources of that chip. Buying 4 Sonoff Basic is the better solution.

closed 2018-12-02 by RavenSystem