RavenSystem/esp-homekit-devices #99: LockMechanism

Partizan7676, 2018-11-26:
Is it possible to add a new device type such as LockMechanism ? Same as garage opener but with adjustable time to switch on and off relay. May be type of lock(elctromechanical or electric). Also good idea to add doorstate sensor too.

RavenSystem, 2018-11-26:
Well, it is possible, but it will not happen soon. Added to [TODO list](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List).

closed 2018-11-26 by RavenSystem

mgoeller, 2019-01-16:
This is my code for a Lock, either an "inching lock" or "continuous lock". Inching lock is suited for a door opener, where the relay needs to be closed for a priorly specified amount. A continuous lock toggles between the states LOCKED and UNLOCKED and stays there until you toggle again.

`
/* V A R I A B L E S */
const int lockType = 0;  // 0=inching lock, 1=continous lock
const int inchingLockTime = 4500;  // inching lock time in ms
const int relay_gpio = 4;
const int led_gpio = 13;
enum {UNLOCKED, LOCKED};
static ETSTimer oc_timer;
static void reach_target_fn();

/* H O M E K I T   C H A R A C T E R I S T I C S */

// predefinitions of variables and functions
homekit_value_t getter_current_state();
homekit_value_t getter_target_state();
void setter_target_state(homekit_value_t value);

// lock target state (locked => 1 / unlocked => 0)
homekit_characteristic_t lock_target_state = HOMEKIT_CHARACTERISTIC_(
    LOCK_TARGET_STATE, LOCKED, .getter = getter_target_state, .setter = setter_target_state);

// lock current state (0 locking/unlocking, 1 secured/unsecured, 2 jammed, 3 unknown)
homekit_characteristic_t lock_current_state = HOMEKIT_CHARACTERISTIC_(
    LOCK_CURRENT_STATE, LOCKED, .getter = getter_current_state);

homekit_value_t getter_current_state() {
    return lock_current_state.value;
}

homekit_value_t getter_target_state() {
    return lock_target_state.value;
}

void setter_target_state(homekit_value_t value){
    if (value.format != homekit_format_uint8){
        printf("Invalid value format: %d\n", value.format);
        return;
    }
    lock_target_state.value = value;
    reach_target_fn();
}

void homekit_notify() {
homekit_characteristic_notify(&lock_target_state,HOMEKIT_UINT8(lock_target_state.value.int_value));
homekit_characteristic_notify(&lock_current_state,HOMEKIT_UINT8(lock_current_state.value.int_value);
}

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "Lock");

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_door_lock, .services = (homekit_service_t *[]){
      HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t *[]){
        &name,
        HOMEKIT_CHARACTERISTIC(MANUFACTURER, "PoWAG"),
        HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
        HOMEKIT_CHARACTERISTIC(MODEL, "Basic"),
        HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1.6"),
        HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
        NULL}),
        HOMEKIT_SERVICE(LOCK_MECHANISM, .primary = true, .characteristics = (homekit_characteristic_t *[]){
          HOMEKIT_CHARACTERISTIC(NAME, "Shelly Lock"),
          &lock_current_state,
          &lock_target_state,
          NULL}),
      NULL}),
    NULL};

void create_accessory_name()
{
    uint8_t macaddr[6];
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    int name_len = snprintf(NULL, 0, "Lock-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len + 1);
    snprintf(name_value, name_len + 1, "Lock-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);
    name.value = HOMEKIT_STRING(name_value);
}

/* L O C K   C O D E */

void relay_write(int state) {
    gpio_write(relay_gpio, !state);
}

static void reach_target_fn() {

    printf("reach_target_fn start\n");

     //continuous lock
    if (lockType) {
        printf("Continous Lock: %d\n", lock_target_state.value.int_value);
        relay_write(lock_target_state.value.int_value);
        lock_current_state = lock_target_state;
        homekit_notify();
    }
     //inching lock
    else if (lock_target_state.value.int_value == UNLOCKED) {
        // unlock
        printf("Inching Lock: %d\n", lock_target_state.value.int_value);
        relay_write(lock_target_state.value.int_value);
        lock_target_state.value.int_value = UNLOCKED;
        homekit_notify();

        // delay
        sdk_os_timer_setfn(&oc_timer, NULL, NULL);
        sdk_os_timer_arm(&oc_timer, inchingLockTime, 0);
        sdk_os_timer_disarm(&oc_timer);

        // lock
        lock_target_state.value.int_value = LOCKED;
        lock_current_state = lock_target_state;
        printf("Continous Lock: %d\n", lock_target_state.value.int_value);
        relay_write(lock_target_state.value.int_value);
        homekit_notify();
    }

}

/* B O O T   I N I T   F U N C T I O N S */

void relay_init() {
    gpio_enable(relay_gpio, GPIO_OUTPUT);
    relay_write(lock_target_state.value.int_value);
}

`