RavenSystem/esp-homekit-devices #102: Sonoff Basic new funktions

yura2507, 2018-11-28:
Hi, Thank You very much for executed work, you helped me a lot to solve different questions.
I kindly ask you to help me to realize the functions in the device Sonoff Basic:
1)	Possibility to change the time for scanning of temperature sensor. For making of fast and precise  thermostat. 
2)	Possibility of connection to GPIO14 an additional relay according to the same scheme of present ( available) relay (through transistor). For realizing in Sonoff Basic 2CH (like Sonoff Dual). (And temperature sensor GPIO4:) )

Thank you in advance.

RavenSystem, 2018-11-29:
1. It's possible. Added to [TODO list](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List)
2. Not possible. It is a very custom hardware setup.

RavenSystem, 2018-11-30:
Try version 0.5.2.

closed 2018-12-02 by RavenSystem