RavenSystem/esp-homekit-devices #104: Install not complete?

Idontknowme-18, 2018-11-28:
After flash is completed and OTA setup is taking place, I can connect to the unit to input wifi details and ota repo etc - but once I hit join, nothing else happens?

After 6 minutes i dont get the LED flash, and i cant pair it in homekit. I can see the device is connected to my router but it does nothing else?

I have tried to reset the device by holding the button for 10 seconds, but the LED doesn't flash and it doesn't appear as a wifi spot again.

Any help would be greatly appreciated, thank-you.

RavenSystem, 2018-11-28:
Try to download files again, erase flash and reflash them.

Idontknowme-18, 2018-11-28:
Thanks for the reply, no good unfortunately.

Re-download otaboot.bin rboot.bin blank_config.bin, erased then flashed, still the same issue?

Once flashed and powered on 230v the SSID is LMC-xxxxxx - is this correct?

Thanks.

RavenSystem, 2018-11-28:
Try this otaboot.bin: 
https://github.com/HomeACcessoryKid/ota/releases/download/0.1.0/otaboot.bin

Idontknowme-18, 2018-11-28:
Same again :( - Only difference that time is it's SSID was OTA-xxxxxx that time.

RavenSystem, 2018-11-28:
Are you using an iPhone/iPad to setup it? In that case, try to setup it using a Mac/PC.

Idontknowme-18, 2018-11-28:
Update - its working! Maybe i wasn't patient enough!! 

Thanks very much for your help.

closed 2018-11-28 by RavenSystem