RavenSystem/esp-homekit-devices #105: Wiki: no blank_config.bin can be found when clicking the file name

Open-Elec-RU, 2018-11-29:
In the Wiki instructions:
Clicking ota_boot.bin conveniently starts downloading the file right away
Clicking rboot.bin takes you to a page where you have to click another Download button
Clicking blank_config.bin in the Wiki leads to some page with no file to download and no Download button.
Is it possible to do all three files like ota_boot.bin - that would be convenient.
(This is on MacOS)

RavenSystem, 2018-11-29:
All files has been fixed in the Wiki.

Many thanks for your feedback.

Moskvitch, 2018-11-29:
Great, thank you.  I actually used the blank_config file from my previous download v 0.4.x - could that be related to the problems I am having with S20s on 0.5.1?  Probably not, but just maybe?

Moskvitch, 2018-11-29:
Please see my comments to #91

closed 2018-11-29 by Moskvitch