RavenSystem/esp-homekit-devices #108: 18b20 temperature sensor

stepanyurkiv, 2018-12-01:
18b20 temperature sensor, temprature is recognized well, only after some time the device is not seen in homekit (Version 0.5.0; 0.5.1; 0.5.2) in modes    5. Thermostat;   6. Switch Basic + TH Sensor; 9. Socket + Button + TH Sensor.

accessoremoto, 2018-12-01:
I have same problem with 18b20 and firmware 0.5.2

RavenSystem, 2018-12-01:
How much time is necessary to get the issue?

stepanyurkiv, 2018-12-01:
As soon as I choose one of the modes, sometimes longer, but after some time it becomes invisible in the homekit. 18B20 soldered permanently and other modes in which there is no temperature measurement is good.

stepanyurkiv, 2018-12-02:
https://drive.google.com/file/d/1ogiFAZkTc7b3oypwDsPPYgiDbLMF-b9N/view?usp=sharing

stepanyurkiv, 2018-12-02:
Already works an hour. OK! I watch next.

closed 2018-12-02 by RavenSystem