RavenSystem/esp-homekit-devices #115: Sonoff Basic New Board Layout

pandarek, 2018-12-04:
"GPIO14 is not inuse anymore. Instead you can use GPIO2 labled as IO2 on the board."
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic

Can you implement the GPIO14 or GPIO2 selection configuration for the new sonoff? The idea is that GPIO2 has the same functionality (toggle) as GPIO14.

More info and photo:
https://github.com/arendst/Sonoff-Tasmota/issues/4515

RavenSystem, 2018-12-04:
Of course.

pandarek, 2018-12-05:
A similar situation will be with TH Sensor and Sonoff Basic without GPIO14

RavenSystem, 2018-12-06:
Added to [TODO List](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List).

RavenSystem, 2018-12-19:
@triak3ak if you want help to install Tasmota firmware, ask on Tasmota web.

RavenSystem, 2018-12-20:
@pandarek try v0.5.5 for new board layout.

wroblewski6, 2018-12-27:
So for this moment if I have new sonoff basic I am not able to use it with your firmware with wall switch connected to gpio14 and gnd?? And there is no option to change It to gpio02 at this moment in your setup ??

closed 2019-01-27 by RavenSystem