RavenSystem/esp-homekit-devices #116: sonoff garaje

sercitecotf, 2018-12-04:
buenas tardes 
muy buen trabajo, muchísimas  gracias :-) 
he configurado un sonoff sv 12v para una puerta de garaje, tengo todos los sensores en off y he puesto 90 segundos que es el tiempo que tarda la puerta en cerrar, pero el sonoff no pasa de abierto a cerrado, se queda todo el tiempo en abierto. he probado varias configuraciones y no lo consigo.

RavenSystem, 2018-12-05:
¿Qué tipo de configuración tienes? ¿Usas dos sensores, uno para cerrado y otro para abierto; o sólo uno para cerrado? ¿Son NO o NC?

sercitecotf, 2018-12-05:
lo estoy usando sin sensores aun no se cuales poner, es en una puerta corredera de entrada a una finca

RavenSystem, 2018-12-06:
Sin sensores es normal que no funcione, ya que el estado de la puerta lo definen los sensores. Como vas a usar un SOnoff SV, te recomiendo usar los dos sensores (puerta abierta y cerrada) e ignorar el tiempo de apertura.

RavenSystem, 2018-12-06:
Te valen cualquiera, ya sean NC o NO, puesto que lo puedes configurar.

Estos son un par de ejemplos:
https://es.aliexpress.com/store/product/10PCS-Wired-Window-Magnetic-Contact-Sensor-Detector-Switch-for-GSM-Home-Alarm-Security-Plastic-Reed-Contact/1188529_32684706524.html
https://es.aliexpress.com/store/product/Wired-Garage-Metal-Roller-Shutter-Home-factory-Gate-Door-Sensor-detector-Magnetic-Contact-Switch-Alarm-for/233325_32820046748.html

sercitecotf, 2018-12-06:
Ok, muchas gracias

closed 2018-12-11 by RavenSystem