RavenSystem/esp-homekit-devices #118: Stateless switches don’t work

guagsak, 2018-12-05:
First of all, thank you for your awesome firmware. 

I have flashed S22 and Sonoff Basic with 0.5.1 version of binary file and updated firmware to the 0.5.4 via OTA. 

I chose the device type 9 with a button option. However, when I choose the other devices associated with Single, Double, and long press button, the targeted device not response. 

When I looked at the stateless switchs in home app, it is blinking as expect as I pressed the button but the target devices not fire. 

Anyone has same experience as mine or know how to set it up. 

Thanks in advance.

RavenSystem, 2018-12-05:
Have you an accessory hub (AppleTV or iPad configured as Home hub)?

guagsak, 2018-12-05:
Yes. I used Apple TV.

RavenSystem, 2018-12-06:
If you see it blinking in Home App, it should work. I have no idea why it isn't working.

dwhp, 2018-12-20:
I have also experienced this. Mode 9 works (single and double), other ones 10 and 3 don’t. You see the single and double press register in home app but it doesn’t do any set actions.

RavenSystem, 2018-12-20:
I have 2x Sonoff S20 (Dev type 3), a S22 (Dev type 9) and a ESP01 (Dev type 10) and I haven't any issue with stateless button. I use them to control Koogeek bulbs (brightness and colors).

closed 2019-01-27 by RavenSystem