RavenSystem/esp-homekit-devices #119: Motion sensor

yusufd1, 2018-12-05:
Hi RavenSystem can you perhaps do an enhancement whereby GPIO 14 can be exposed to the home app as a motion sensor. Motion sensors are currently available that work with HomeKit. Basically it’s a simple NO or NC type switch that is an icon that simply lights up on the home app and sends the user a notification. A pir motion sensor can then be connected to the sonoff GPIO14 or any other type switch whereby the user requires a notification as soon as the switch is closed. I have an electric fence and if the alarm is activated I get the motion sensor notification on the home app. The code is already written for the home app and is available on GitHub. There is also a YouTube video by the user Quickpi. The current way I am using it is via a wemos d1 mini and a raspberry pi. It will be great if I can just simply deploy one ravencore sonoff that can control the power to the electric fence as well as alert me when the alarm is triggered via GPIO14. 

I hope you will consider this enhancement as it can be very useful in many different scenarios where users need a sonoff and also require to be notified.

RavenSystem, 2018-12-05:
I know how HomeKit motion sensor works, but I don't know exactly what you mean. I setup GPIO14 as input pin and this triggers a HomeKit Motion Sensor, that is ok; but what happens with relay? how relay works?

yusufd1, 2018-12-06:
Hi setting up gpio 14 as input pin that triggers motion sensor on home app will be perfect. Don’t worry about relay.

RavenSystem, 2018-12-06:
I'm going to develop a more complex device, including a motion sensor, alarm, occupancy, etc...

Added to [TODO List](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List)

yusufd1, 2018-12-06:
That will be excellent. Thank you for your help. Most home alarms including mine are activated and deactivated via a momentary switch so if you are able to write code whereby the sonoff relay toggles on and off with every single press on the home app it would be perfect. The home app should maintain the state as on or off per single press although the relay will toggle quickly from open/closed back to open. Exactly how the LED toggles right now on ravencore per press. I can use the LED GPIO for this implementation right now but as you know when there is a reboot then the LED GPIO blinks a few times. This will cause my alarm to activate/deactivate etc so hence I have not done it.

claire078, 2019-01-05:
I could really welcome this feature, I would use it by my alarm system to send me a notification when I have an alarm at home. 
I can’t wait to test it. 👍

i3laze, 2019-01-05:
Just wondering, which hardware sensors would you use for occupancy?
There are lots of PIR’s around for motion.. but sensing volume is something different.

Want to control garage parking lot.

maccoylton, 2019-01-06:
I have a motion sensor based on the HC-SR5012 here :- https://github.com/maccoylton/esp-homekit-motion-sensor
 I've no issues with False alerts, with the latest module I bought, I did see this on previous modules.  

Use the firmware or this version if you want to look at the code:- 
https://github.com/maccoylton/esp-homekit-motion-sensor/commit/3753ca2bc21666734bb0a49cb4728c021b01f310

i3laze, 2019-01-06:
Still all of it is about motion (as this issue thread is). What about Occupancy?

These sensors mentioned above cannot distinguish an object or person present, but not moving.

maccoylton, 2019-01-06:
I use the automation in the home app along with my motion sensor to turn on the kitchen lights but only if someone is Home.  This of course works on the presence  or not of the family members with their iPhones

AramVartanyan, 2019-01-06:
HC-SR501 is little bit tricky.
The cables connecting the sensor and the esp chip input pin must be soldered. The false alerts are due to the low working current (nA). Also it is very important to put a 10kOhm pull-up resistor to the signal line.
After those small measures the false alerts are reduced to Zero.

peros550, 2019-01-06:
> HC-SR501 is little bit tricky.
> The cables connecting the sensor and the esp chip input pin must be soldered. The false alerts are due to the low working current (nA). Also it is very important to [...]

Hi I'm trying to understand the logic. Are you sure about the pull up resistor? The sensor gives HIGH when it finds movement. Having a pull up would between the input/signal pin and Vcc (up) would constantly give HIGH in the input pin, am I correct?

Perhaps a pull down between between input pin and the GND would be a better choice?

RavenSystem, 2019-01-06:
I suggest to use my advanced button library to avoid false positives and interferences:
https://github.com/RavenSystem/esp-homekit-devices/tree/master/libs/adv_button

Example:

```
void movement_detected_fn(const uint8_t gpio) {
     printf("Movement detected\n");
}

void movement_not_detected_fn(const uint8_t gpio) {
     printf("Movement not detected\n");
}

adv_toggle_create(SENSOR_GPIO, false);  // false -> without internal pull-up resistor
adv_toggle_register_callback_fn(SENSOR_GPIO, movement_detected_fn, 0);    // Low gpio state
adv_toggle_register_callback_fn(SENSOR_GPIO, movement_not_detected_fn, 1);    // High gpio state

```

i3laze, 2019-01-07:
Found out there are bunch of ultrasonic sensors for occupancy/range detection.
Like HC-SR04 or JSN-SR04T (outdoor edition).

But.. there is some math calculation required for it to work.
Distance is 0.25-3m.
How about supporting this one for occupancy?

peros550, 2019-01-09:
> I suggest to use my advanced button library to avoid false positives and interferences:
> https://github.com/RavenSystem/esp-homekit-devices/tree/master/libs/adv_button
> 
> Example:
> 
> ```
> void movement_detected_fn(const uint8_t gpio) {
>      printf("Movement detected\n");
> }
> 
> void movement_not_detected_fn(const uint8_t gpio) {
>      printf("Movement not detected\n");
> }
> 
> adv_toggle_create(SENSOR_GPIO, false);  // false -> without internal pull-up resistor
> adv_toggle_register_callback_fn(SENSOR_GPIO, movement_detected_fn, 1);    // High gpio state
> adv_toggle_register_callback_fn(SENSOR_GPIO, movement_not_detected_fn, 0);    // Low gpio state
> ```

Works like a charm (by swapping output states for my sensor) ! 
Thank you! @RavenSystem 

Anyone interested should give it a try. 

In regards to the false triggers I noticed another factor that could negatively affect the result. The power supplier. If the power supply is not good enough then the HC-SR505/HC-SR501 does not get the required voltage (4.5-5V) and does not work properly.

closed 2019-01-27 by RavenSystem