RavenSystem/esp-homekit-devices #123: Sonoff T1 R2 UK reboots every 60 seconds

WizBangCrash, 2018-12-08:
Hi,
I have been using RavenCore for a while with the multiple Sonoff Basic units in my house. Last week I bought a Sonoff T1 and have re-flashed with RavenCore but I cannot get it to pair with HomeKit.
Having spent some time debugging the issue it appears that after the unit has been configured on the network and the ravencore firmware has been downloaded OTA the unit reconnects to my WiFi, stays up long enough to advertise it's presence to HomeKit, but reboots before the device can be added in the HomeKit app.

The sequence I follow is:
1. Erase the flash (over serial)
2. Flash the roboot.bin, otaboot.bin and blank_config.bin from the v0.4.10 release (over serial)
3. Wait for a couple of mins and then connect to device over WiFi
4. Configure my WiFi settings and set the repository to `RavenSystem/ravencore` and OTA binary to `main.bin` before clicking "Join"
5. Wait until the OTA process completes (normally 3-7 mins)

After these steps I see the device in HomeKit when I try to add an accessory, but if I monitor my router logs I can see that the Sonoff unit is only staying connected to the Wifi for about 60 seconds and then disconnects. It reconnects about 10 seconds later and then repeats this cycle.

If I watch the sonoff device I see the following:
1. WiFi LED lights for 2 seconds then goes off for 2 seconds
2. WiFi LED flashes for 0.5 sec
3. Left touch sensor turns blue and stays blue
4. 40-50 seconds later left touch sensor turns off and I go back to step 1.

Any advice on how to resolve this would be most welcome :-)

My Sonoff is:
[image]

RavenSystem, 2018-12-08:
I don't know why your device is restarting every 40-50 secs. Maybe a faulty unit. Can you provide logs, please?

WizBangCrash, 2018-12-09:
Thanks for the response. Can you point me in the right direction for some notes on how to get a copy of the logs?

RavenSystem, 2018-12-11:
Sure, you can use `screen` command after connecting your device to computer:
`screen /dev/<your_device> 115200`

WizBangCrash, 2018-12-11:
Thanks. I'll get some logs to you as soon as I can.

WizBangCrash, 2018-12-12:
Hi @RavenSystem,
I have used the `screen` command to capture a few iterations of the rebooting sequence.
I've had a look at the results and don't yet know enough about the SDK to understand what is going wrong, but you can clearly see a message on line 79 showing that the SonOff has connected to my WiFi network (`connected with MyWiFiNetwork, channel 9`) and got an IP address (line 81) and then in line 85 we get a message stating `wifi_config: wifi_config_station_connect: No configuration found`

I have followed the same setup procedure with this device as I have with the Sonoff Basic's I have previously configured.

Any help you can give would be most welcome.

[screenlog.txt](https://github.com/RavenSystem/esp-homekit-devices/files/2673872/screenlog.txt)

WizBangCrash, 2018-12-12:
I've also uploaded an additional log of me connecting to the RavenCoreXXXXXX SSID and setting the WiFi parameters.
You can see me update the WiFi settings at line 77. The device then reboots, connects to my wifi at line 129 and states it has successfully connected on line 148.
The HomeKit server is then started (line 156).
On line 177 I get the following: `RC >>> Resetting device to factory default` just after an mDNS announcement :-(

I hope this helps in getting to the bottom of the issue.

[screenlog2.txt](https://github.com/RavenSystem/esp-homekit-devices/files/2673941/screenlog2.txt)

RavenSystem, 2018-12-13:
Is your GPIO0 released?

WizBangCrash, 2018-12-15:
Yes.
I had soldered a wire to on to R19 to hold GPIO0 low during the re-flashing process, but I desoldered it write after flashing and have checked that there is nothing holding it low.
I am finding that the device just resets a random amount of time after booting.
I have found that now I can monitor the logs I am able to time things so that I can get all the way through to pairing the device to HomeKit and then anything from a few seconds to a minute or two after the first mDNS broadcast the device resets.

RavenSystem, 2018-12-15:
Device keep doing factory reset?

WizBangCrash, 2018-12-16:
FYI I flashed the device with the Tasmota firmware today as I was concerned it was a hardware related issue with the Sonoff unit.
Tasmota software loaded and worked fine. No resets.
If I get time this week I will look at building your firmware locally and adding some debug code to try and get to the bottom of the issue.
Have you ever used VSCode and Platform to build RavenCore? I'm familiar with this and wondered if you had already tried it?

RavenSystem, 2018-12-16:
Sorry, I forgot GPIO that performs factory reset is GPIO10 in Dual type, not GPIO0. Check status of GPIO10. 

I use Xcode for coding, and gcc to compile.

WizBangCrash, 2018-12-16:
GPIO10 (pin 19) is connected directly to an input pin on an 8051 device that controls the LEDs on the touch pads and the main LED.

RavenSystem, 2018-12-18:
Sorry, another mistake. I have read all your logs, and problem is with GPIO-0 for sure.

Partizan7676, 2019-01-11:
Can someone explain which type of T1 supported? Only 3ch? Or I can use 1 and 2 ???

RavenSystem, 2019-01-11:
1ch -> dev type 1.
2ch -> dev type 2.

closed 2019-01-27 by RavenSystem

pmstmb, 2019-03-04:
I have the same problem. Any solution?

WizBangCrash, 2019-03-04:
Hi @pmstmb ,
I never did get to the bottom of this. I couldn't set up my own debug environment and @RavenSystem closed the issue. I ended up switching to Tasmota (https://github.com/arendst/Sonoff-Tasmota) as it had no issues with the T1 R2.
I'd still like to switch back to RavenSystem at some point though as I'd prefer to have just a HomeKit environment and not need an MQTT server.
I may come back to it in a few months when I've finished my current project.

pmstmb, 2019-03-06:
Hi

Thank you for your replay. I hope, in future will be possible to use ravencore for sonoff touch. It is better, that is not neccessary to use bridge. I flashed sonoff touch with tasmota now, it works perfectly, but only I have to use bridge for homekit.

PauloVmm, 2019-07-21:
my sonoff t1 with 3 switches, is running just a switch on homekit, could anyone help me?