RavenSystem/esp-homekit-devices #129: FlashEZ / Esp Easy for ravencore?

seritos, 2018-12-11:
Hi All,

Got a new version of sonoff S20. Is it possible to upgrade it with FlashEZ or EspEasy to directly flash it? I would use the main.bin

Any other option for windows?

Ezkerrik!

RavenSystem, 2018-12-12:
I don’t use Windows, only macOS and GNU/Linux, but you can use any program ready to flash ESP8266 chips, simply setting memory adresses for each file, like the esptool command.
0x0 for boot.bin
0x1000 for blank_config.bin
0x2000 for otaboot.bin (or main.bin if you don’t want to use OTA updates)

seritos, 2018-12-12:
Thanx Jose for your fast reply. 

I am also a Mac User. How would you recommend it to flash?

yukonx, 2018-12-12:
**setup**
[https://upload.cc/i1/2018/12/12/fRrUku.png](https://upload.cc/i1/2018/12/12/fRrUku.png)

**ESPFlashDownloadTool_v3.6.4**

https://drive.google.com/open?id=1b-5jma9HZzI_JZqZtCogz1hp3n6ejuZT

seritos, 2018-12-12:
Thank to both of you.

Just flashed perfectly with ESPFlash in windows and with esptool in Mac. 

Works flawlessly!

closed 2018-12-12 by RavenSystem