RavenSystem/esp-homekit-devices #130: Temporary "No response" status

peros550, 2018-12-12:
I get this very strange behavior. It's not very important, because the issue resolves by itself.  I'm reporting it in case anybody else is experiencing the same. 

Sometimes, when I open the home app, some of my devices report "No response" status. This is not very important as they device becomes available again after 30 seconds or less. I am just wondering if there is something to fix or if this is isolated situation in regards to my own setup (Asus Router etc). 

Thanks

RavenSystem, 2018-12-12:
I have an Asus RT AC5300, and I have not any issue since I use Asus Merlin firmware. All devices respond very very quickly.

RavenSystem, 2018-12-12:
Asus Merlin firmware:
https://asuswrt.lostrealm.ca/home

peros550, 2018-12-12:
Hi Jose, 

thanks for your quick feedback. Again, this is a minor issue, if I could call it an "issue" after all :)

Yes, I use the Merlin firmware in two routers (RT-AC68U) which are located in two different houses. I have this randomly unavailable status sometimes. Maybe I should try changing the Wifi channel for start. If you have any other suggestion on the advanced Wifi settings , let me know :) 

I'll let you know how it goes. 

As I said, I was interested to know if anybody else is experiencing the same. Your firmware starts to attract more and more people which is good thing and evidence of the high quality software that you build. 

Thank you

RavenSystem, 2018-12-12:
@peros550 try setting Bluetooth coexistence in pre-emptive, under tab Professional in Wireless section. In my case, that was the trick.

Gio1978, 2018-12-13:
I have the same problem.

seritos, 2018-12-15:
I am experiencing the same “ issue”. Its true that it fixes by restarting the homekit app.

RavenSystem, 2018-12-18:
@peros550 FYI This is my 2.4GHz config:
[image]

peros550, 2018-12-23:
Hi Jose, 
I mirrored your settings but didn't have any positive development. Problem still remains. I'm wondering if this issue could have its root into a combination of factors. 

For example, one thing I've noticed when playing with my router's wifi settings is that  Sonoffs running Ravencore firmware do not always reconnect to wifi after a wifi disconnection.  

Have experienced/tested this scenario in your setup?

RavenSystem, 2018-12-24:
@peros This is my complete wifi setup with my Asus GT-AC5300, used to manage more than 10 wifi devices (I actually have about 30 wifi devices). Try it:

- General Wifi Setup:
[image]

- 2.4GHz Professional tab:
[image]

- 5GHz-1 and 5GHz-2 Professional tabs (both have the same settings):
[image]

- Smart Connect Rules:
[image]

gerarddvb, 2018-12-24:
Same here, connection issues start after 2-3 days. I have 2 Sonoff and a D1, they all have same problem

RavenSystem, 2018-12-25:
@peros550 have you tried latest AsusMerlin settings?

peros550, 2018-12-25:
Hi Jose, 

yes I mirrored every setting I could (some settings are not present in my model). So far so good. 25 hours since last router reboot and everything looks good. 
I'll be monitoring for a few more days.  

My next test (in a few days, assuming all 3 Sonoffs will keep responding until then) will be to reboot router and see if all Sonoffs reconnect. 

Thanks & Merry Christmas

peros550, 2018-12-26:
Good day Jose!

I would like to give you an update. Today this morning, all my 3x Sonoffs gave the red "no response" message. I thought that was very strange and unusual as the reason I first opened this issue was for random, non permanent, "no responses" of some of my devices not all of them. 

A few days ago, I have started testing Maxim's homekit firmware ( the temperature sensor example) on a Wemos D1 mini. This Wemos D1 was also in "no response" status. I decided to power reset only the Wemos. The Wemos became available again in the Home app and after a few minutes so did the Sonoffs. 

It looks there is some kind of interference between these devices running Homekit capable firmware. 
Perhaps the mDNS service is somehow affected by other devices? I am only making assumptions... I would love to have your view on this. 

I will keep on monitoring and let you know how it goes.

RavenSystem, 2018-12-26:
A faulty mDNS service can accept to others because these devices concatenate mDNS responses. I saw your comments on another issue in esp-homekit-demo repo, but they are talking about esp32, and Wemos D1 mini is an esp8266. Maxim's HomeKit framework use different mDNS implementations fo each chip: esp32 uses the official Espressif SDK mDNS, and esp8266 uses an improved version from esp-open-rtos.

I think that your problem is the HomeKit framework version you are using. HomeKit version from esp-homekit-demo is very old, and maybe your esp-open-sdk is not old. If you clone my repo, you will get a modified and stable version of HomeKit framework (focused on stability for day use), and a well tested esp-open-rtos version. You can move your code to devices directory of my repo, and copy the Makefile from RavenCore.

peros550, 2018-12-26:
Thank you for the explanation Jose. I saw a guy talking about a potential fix and wanted to try. 

I had thought about that (old version of esp-homekit) and tried to manually bring to latest version by running the following: 

```
cd submodule_name
git checkout master && git pull
cd ..
git add submodule_name
git commit -m "updating submodule to latest"
```
I 'll try you suggestion for sure though. Many thanks for your support.

By the way, I remember you were interested on having IR capabilities. Here is a guy who achieved to send successfully IR commands.  
https://github.com/pbendersky/ir-esp-uart

Will keep you updated

peros550, 2018-12-31:
Hi Jose, 

here are some updates from my side: 
a)  on the random "no responses": It is 7 days since I last rebooted my router having applied your proposed router settings. All my 3 devices are responsive again for this time period. I'm pretty positive they will not disconnect again. I'm going to monitor for another week and let you know. After that I'm going to cause a wifi disconnect and see if they reconnect again. 

b) On my custom code, I did what you said. I used your already set up repository as a basis and moved my code there. It is already a few days and my Wemos D1 mini is responsive. Using Maxim's repo the "no response" status would come up very quickly. 

c) On the IR library I posted above. I have tried it and it works nicely. Here is an issue that I first opened myself, but then closed it providing instructions on how I used the library. 
https://github.com/pbendersky/ir-esp-uart/issues/1

Hope the above information will be helpful

claire078, 2019-01-05:
Hi,
I have this problem too. I had it before too but since 0.5.2 it was super. Today I updated to 0.5.7 and have this problem again. Not often as before but there’s. 
Could you check the code again? It could be AC interference problem?
Thanks.  🙏

peros550, 2019-01-05:
> Hi,
> I have this problem too. I had it before too but since 0.5.2 it was super. Today I updated to 0.5.7 and have this problem again. Not often as before but there’s.
> Could you check the code again? It could be AC interference problem?
> Thanks. 🙏

When your device shows "no response" do you need to reset it to come back as normal or it gets back to normal by itself? If your situation is the latter, roughly for how long the no response situation lasts?

claire078, 2019-01-05:
It gets back to normal about 10 secs later. 
Before this time was more, half minute or more. 
But after 0.5.2 I saw never this “no response”.

RavenSystem, 2019-01-06:
Anyone was using device type 1 with gpio14 toggle enabled? Now this has changed, and there was a bug in 0.5.7 with that and devices using gpio14 toggle crashed. This is solved with 0.5.8. Please, read Wiki to configure gpio14 toggle, now called external toggle 1.

peros550, 2019-01-07:
For anyone who didn't install 0.5.7 , is there still a need to erase flash to go at 0.5.8 or OTA method can be used?

RavenSystem, 2019-01-07:
Don’t erase, update with OTA.

peros550, 2019-01-07:
Thank you Jose.  
By the way, do you know if the <wifi_config.h> can be instructed to always use a specific Wifi SSID and password?
I'm working on building an A/C virtual accessory to control my A/C using IR and I have to re-flash very often. And I was wondering if I could bypass that step.

peros550, 2019-01-07:
Closing, for the past few weeks I had no further instance of the no response situation.

closed 2019-01-07 by peros550

Partizan7676, 2020-10-15:
Just open it for ask.
Have strange situation. At this moment have 3 HAA devices - 2 sonoff mini and basic r2. One mini works just fine and its connected to Ubiquiti Nano HD. Others connected to different APs and until i reboot router i get response from them just fo few minutes. But if i connect my phone to their AP - they response as well. Any idea?