RavenSystem/esp-homekit-devices #131: Can’t connect to WiFi after setting SSID/password

caligari74, 2018-12-12:
Hi, first of all thank you for the great software!  After flashing and selecting the SSID and password and then waiting 10 minutes or more, my Sonoff Basic never connects to my home WiFi.  I can’t add it to HomeKit.  The router is an Apple Time capsule.  The AP is always visible (Ravensystem-xxxxxxxx).  Should this disappear when connected?  My authentication on the time capsule is set to WPA/WPA2. Any suggestions?

RavenSystem, 2018-12-12:
AP should dissapear when device connects to your wifi. 

Can you provide logs, please?

caligari74, 2018-12-12:
The Time Capsule doesn't have any logs.  Unbelievable.  
I did try to connect it to my iPhone hotspot, and got the same behavior.  I flashed another Sonoff Basic and it behaves the same way.  No connection to WIFI. I haven't had this ever happen with other devices.
Unfortunately I don't have a spare router to try out. 
I'd love to get this working.  Any ideas for things I could try?

RavenSystem, 2018-12-13:
I was talking about logs from your Sonoff.

caligari74, 2018-12-13:
**solved**

After checking the log (via screen) I saw the Sonoff was indeed trying to connect to my WIFI:

> no Above&amp;Beyond found, reconnect after 1s
> reconnect
> scandone
> scandone
> reconnect
> RC >>> Toggle Switch manual
> RC >>> Relay 1 -> 0
> scandone
> no Above&amp;Beyond found, reconnect after 1s
> reconnect
> RC >>> Saving last states
> scandone
> no Above&amp;Beyond found, reconnect after 1s
> reconnect

My WIFI SSID is **Above&Beyond**.  I guess the code doesn't like that.  I changed the SSID name to something else and now it works great.  Maybe a fix will come in a future update?
Thanks very much for your help!  Really great software!

caligari74, 2018-12-13:
Ah boy...the code here won't display what the log showed.  &amp is converted into & the way it should.  Here's a screenshot.
[image]

caligari74, 2018-12-13:
Hope this info helps.  Closing this issue.

closed 2018-12-13 by caligari74