RavenSystem/esp-homekit-devices #132: Temperature and humidity support for S20 or Basic ( no TH)

seritos, 2018-12-12:
Hi,

I have seen that it is possible to add a humidity and or temperature to the BasicTH and S22, but would be possible to add support for the S20 and basic models ( no TH) 
Additionally, the EVE ( or other) app will record the historical temperature measured by the device? 

Thanx a lot!

RavenSystem, 2018-12-13:
You can use a Sonoff Basic with a sensor attaches to GPIO14, that is a TH. 

Historical is not saved.

seritos, 2018-12-13:
Is the device the one that stores the historical? if the eve temperature sensor can do it... why not then, the Sonoff if this is compatible with the Eve app? 

additionally, can I use a S20 and attach it one of the 3 sensor compatible? ( DHT11, DHT22, DS18B20)
Thnx

RavenSystem, 2018-12-18:
> Is the device the one that stores the historical? if the eve temperature sensor can do it... why not then, the Sonoff if this is compatible with the Eve app?

Device is the one that stores the historical, but it is a function that only works into EVE App using HomeKit protocol, but outside Apple HomeKit services. I'm not interested in add that functionality.

> additionally, can I use a S20 and attach it one of the 3 sensor compatible? ( DHT11, DHT22, DS18B20)

You can use any device (S20 included) to attach a sensor to GPIO14.

> last question: what is the resolution config for the 18b20? 9bit? 12 bit?

I don't know. I use library included in esp-open-rtos SDK.

closed 2018-12-25 by RavenSystem

seritos, 2018-12-31:
Just a quick comment... the S20 doesn't have GPIO14... where can I connect then the DHT22 or other temp sensor? 

Thanx