RavenSystem/esp-homekit-devices #133: esp8266 without button

egorist, 2018-12-15:
I installed firmware on esp8266 chip (not sonoff). I use it with 4 relays and device type 4. Everything works excellent. But now I need to add this device to another wifi point. And there is no button on my device. As I see, I can't reboot settings by pressing some gpios on esp8266. so, what should I do?))) 
If I turn my router of after 5 minutes esp8266 make access point automatically. But maybe you could make virtual button in eve app for resetting wifi settings?))
and another question. Is it possible to add temperature or other sensors from devices type 5-8 to device type 4? for example 4 relays, temperature sensor and irrigation sensor? Are there plans to create another firmware for esp8266 with all abilities from ravencore and own builder for gpio customization?

RavenSystem, 2018-12-25:
RavenCore is not focused on custom hardware. Sonoff devices have buttons. If you want to build a custom firmware, I suggest this repo: https://github.com/maximkulkin/esp-homekit-demo

> Are there plans to create another firmware for esp8266 with all abilities from ravencore and own builder for gpio customization?

I haven't enough free time to do that. Remember that this is a hobby ;)

RavenSystem, 2019-01-27:
@egorist A new firmware will be released with an own builder for gpio customization.

closed 2019-01-27 by RavenSystem