RavenSystem/esp-homekit-devices #135: Feature Request : Faster on

rhjensen79, 2018-12-17:
Is it possible to make the on feature, faster after the power comes on ?
Today the delay from power comes on, until the light (og whatever you have connected) is about 2-5 seconds.

It would be great if it was instant (or close), so the Sonoff is invisible, in the setup.

RavenSystem, 2018-12-17:
I understand that your device is set to turn on light on boot, and you want to reduce the delay between device powers on and relay turns on, is it?

rhjensen79, 2018-12-17:
Exactly. Sorry for not being more clear :-)

/Robert

RavenSystem, 2018-12-20:
It id caused by a delayed initial timer, but I want to remove it.

RavenSystem, 2018-12-20:
Try v0.5.5.

rhjensen79, 2018-12-20:
It works. 
Now on is almost instant. 

Thanks.

I'm closing the issue.

closed 2018-12-20 by rhjensen79