RavenSystem/esp-homekit-devices #138: Request: buttons/switches only mode

dwhp, 2018-12-20:
Hi. I must say again your work is great. I was wondering if it was possible to have another mode where an esp8266 board (or moded Sonoff eg. 4ch) can be inputs only ie. no Relays. With options: basic on/off momentary, toggle like gpio14 mode1, single/double/long like your stateless AND ability to enable/disable 2nd, 3rd, etc Switches. A dev board has GPIOs 0,1,2,3,4,5,12,13,14,15,16 available. You may have a better idea, just would like to use some existing matching house toggle switches (1-3gang) purely for controling lights/lamps/sockets etc not wired to switch. Mode 12 doesn’t work, but I did suspect that. Thanks :)

RavenSystem, 2018-12-20:
This firmware is not focused on a very custom hardware setup; if you want to build a firmware for your custom hardware, I suggest you https://github.com/maximkulkin/esp-homekit-demo

Mode 12 doesn't work because actually that device type is unused.

Supporting ifan02 was considered and dropped.

dwhp, 2018-12-20:
Thanks, I forgot about HomeKit-Demo when I became obsessed with your work. Will do.

closed 2018-12-25 by RavenSystem