RavenSystem/esp-homekit-devices #140: Sonoff Basic R2 - Sonoff RF R2 Power V1.0

dwhp, 2018-12-20:
I have unintentionally purchased the new version of the basic.. Everything works fine, however since it now uses the ESP8285 chip, GPIO 14 is NOT available (in fact i did not realized until i was looking for gpio14)  - there is a pad on the board for gpio2 tho. is it possible to please add a "GPIO2 Toggle" in addition to "GPIO14 Toggle"to mode 1 please.. Thanks.

dwhp, 2018-12-20:
Reference: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic

    GPIO0 = KEY
    GPIO2 = IO2 (no pullup)
    GPIO12 = RELAIS
    GPIO13 = LED1

Codkit, 2018-12-20:
Hi
Its just happened to me too ! May i ask you how to flash esp8285? Thanks

dwhp, 2018-12-20:
same as other 8266 sonoff.. "FT232RL 3.3v FTDI Serial Adapter module (usb to ttl)"  i bought off ebay.. before powering on i hold button, plug in usb, flash as per instructions. (I don't let go of button until i unplug usb when flashing, if that means anything.) remove tx/rx and repower with ttl adapter and finish OTA wifi settings etc.. i dont touch/bump it until i see it in home app - 5-15minutes, on the rare occasion that it takes longer - I just try again from beginning 20min+...

dwhp, 2018-12-20:
honestly, i didn't even realize until i was searching for gpio14... i think esptool detects chip

Codkit, 2018-12-20:
Thanks very much, i did flashed it sucessfully , i wait 20mins+ still no go after power it up......

dwhp, 2018-12-20:
Did you successfully put in OTA settings? 

1. Connect all 4 pins from TTL (I free hand so I apply stable pressure)
2. Hold button (I personally don’t let go)
3. Plug in usb
4. Erase 
5. Unplug and plug back in usb
6. Program
7. Unplug usb (I let go of button after this)
8. Remove tx and rx
9. Plug in usb (good stable power pin connection)
10. Using iPhone - connect to ota-xxx AP
11. In login screen enter OTA setting RavenSystem/ravencore AND leave main.bin if latest firmware
12. Choose & enter your WiFi password 
13. Press join. - then don’t anything. Don’t reboot either. 
14. I watch it connect and reconnect etc in my router, and watch up time. It reboots a few times during setup, when I see it was up for over 7mins I’ll check home app, if not there yet I’ll wait another 5-15mim before I try again. Now I have a technique, success is 99%

Codkit, 2018-12-20:
Thanks man , it works if i hold the button while flashing , strange it used to work with 8266 without holding it.

RavenSystem, 2018-12-20:
> I have unintentionally purchased the new version of the basic.. Everything works fine, however since it now uses the ESP8285 chip, GPIO 14 is NOT available (in fact i did not realized until i was [...]

https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

closed 2018-12-20 by RavenSystem

cnmarch6, 2019-02-02:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 493, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 332, in command
    p = self.read()
  File "/usr/local/bin/esptool.py", line 277, in read
    return next(self._slip_reader)
  File "/usr/local/bin/esptool.py", line 1873, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 323, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (6, 'Device not configured')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2959, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2952, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2646, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 259, in detect_chip
    detect_port.connect(connect_mode)
  File "/usr/local/bin/esptool.py", line 460, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "/usr/local/bin/esptool.py", line 440, in _connect_attempt
    self.sync()
  File "/usr/local/bin/esptool.py", line 379, in sync
    timeout=SYNC_TIMEOUT)
  File "/usr/local/bin/esptool.py", line 343, in command
    self._port.timeout = saved_timeout
  File "/usr/local/lib/python3.7/site-packages/serial/serialutil.py", line 368, in timeout
    self._reconfigure_port()
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (6, 'Device not configured')