RavenSystem/esp-homekit-devices #141: Sonoff T1 3ch on 0.5.5 setup issue

poktkm, 2018-12-22:
Did an install to sonoff T1 3 gang US version using OTA method to an  so I assume that I did get the latest version which is 0.5.5. The installation went fine until I try to setup the device by changing the device type to 11 and then restart.

The device didn't come back online and instead the LED is flashing slowly. Any idea?

RavenSystem, 2018-12-22:
Device type 11 is not tested, because I haven't a compatible device, but with your help I can improve it. Can you provide logs, please?

poktkm, 2018-12-22:
I am more than happy to help, but I am no esp expert. Can you tell me exactly how to get the log?

Thank you
Paul

RavenSystem, 2018-12-22:
What operating system are you using?

poktkm, 2018-12-23:
osx

tor5085, 2018-12-23:
Same the problem , os windows 10  i

RavenSystem, 2018-12-25:
@poktkm You can use `screen` command after connecting your device to computer:
`screen /dev/<your_device> 115200`

Attach log in a .txt file, please. Don't copy/paste here.

poktkm, 2018-12-25:
Got it. Please refer to the attached log file, the log is only start just before I change the device type to 11 and restart the device, anything before that such as homekit setup up log is not included because I don’t think that there is any problem there. After changing the device type to 11, the log shown that the program seem to loop back to the same precedure. I waited for it to loop 3 times and stop the log.

PS. I have try changing the devide type to 4 (Switch 4ch) and it seem to work fine, except 4th switch doesn’t register to anything on the physical 3ch switch. This may help you reusing the code somehow.

Have Fun
Paul

> 
> @poktkm <https://github.com/poktkm> You can use screen command after connecting your device to computer:
> screen /dev/<your_device> 115200
> 
> Attach log in a .txt file, please. Don't copy/paste here.

Script started on Wed Dec 26 00:02:19 2018
[?1034hbash-3.2$ exitscript screen.logeen /dev/tty.usbserial-14330 115200
[?1049h[!p[?3;4l[4l>[4l[?1h=[0m(B[1;24r[H[2J[23B[7mCannot open line '/dev/tty.usbserial-14330' for R/W: Resource busy[27m  
[67C[1K[H[23B[7mSorry, could not find a PTY.[27m  
[29C[1K[H[?1l>[24;1H
[?1049l[screen is terminating]
bash-3.2$ screen /dev/tty.usbserial-14330 115200
[?1049h[!p[?3;4l[4l>[4l[?1h=[0m(B[1;24r[H[2J[H[2J$�abQ�aׅ�aaű�b�RC >>> Saving settings
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with B8FBF993-DC7A-4107-B32C-9BC65021323F
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
RC >>> Saving settings
>>> HomeKit: [Client 4] Update Characteristics
RC >>> Restarting device
rm match
del if0
usl
sul 0 0
!!! HomeKit: [Client 4] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 4] Closing client connection

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
[... 241 lines cut ...]
frc2_timer_task_hdl:0x3fff5348, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 20 2018 12:45:42
phy ver: 273, pp ver: 8.3

RC >>> RavenCore firmware loaded
RC >>> Developed by RavenSystem - José Antonio Jiménez
RC >>> Firmware revision: 0.5.5

RC >>> Loading settings
RC >>> Loading device_type -> 11
RC >>> Loading saved states
RC >>> Initializing hardware...
RC >>> Hardware ready
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to Horry
mode : sta(84:0d:8e:48:5a:5e)
add if0
scandone
[?1l>[24;1H
[?1049l[detached]
bash-3.2$ exit
exit

Script done on Wed Dec 26 00:03:54 2018

RavenSystem, 2018-12-25:
Many thanks. A fix is on the way.

poktkm, 2018-12-26:
Working now!!! Thank you very much

Happy boxing day and Happy New Year

closed 2018-12-26 by RavenSystem

mdoliveira10, 2019-06-07:
Bom dia 
estou procurando ajuda fazer o sonoff T1 funcionar com o homekit apple? alguem pode me ajudar?