RavenSystem/esp-homekit-devices #142: Request: Use humidity to trigger relay

moenster, 2018-12-22:
When selecting device type 5 on a TH, it would be nice, if you could use humidity to trigger the relay. In my case specifically, I want to use humidity to trigger if the relay should start a ventilator. It could be a new device type but my point is to use humidity to trigger a fan and not temperatur.

RavenSystem, 2018-12-25:
You are asking for a HOMEKIT_SERVICE_HUMIDIFIER_DEHUMIDIFIER.

digiservice2001, 2018-12-26:
The Humidifier/Dehumidifier accessory is very useful. I hope, You will make it asap... 😀

Btw, many many Thanx for Your excellent work! 👍

moenster, 2018-12-26:
> You are asking for a HOMEKIT_SERVICE_HUMIDIFIER_DEHUMIDIFIER.

Yes, that would cover what I need, and then some.
Thanks a lot, for considering it.

RavenSystem, 2018-12-26:
Ok, ok, added to TODO List: 
https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

RickNunn, 2019-01-15:
+1 to wanting this

closed 2019-01-27 by RavenSystem