RavenSystem/esp-homekit-devices #143: Request: Garage Door relay state time

Korcsog, 2018-12-23:
Hi!

Is it possible to make a new Characteristic for changing the opening/closing relay time? The 250ms is too short for my garage door to recognize the command. Compiled the non-OTA version with some modification, in my case 1000ms is the minimum time for the relay to be pulled for proper opening/closing. It would be good to have some options like 250/500/1000ms or any custom time for the relays to be pulled. 
Others might have this problem either! 😄

archiolochos, 2018-12-23:
Hi to all!

I also have got the problem, that my garage-switch (the relay) needs to be always closed and only if i want to trigger the opening or closing it needs to open the relay for a short time. so it means i would need a "inverted" relay behaviour. Would be great if it is possible to implement that.

Merry Christmas everyone!

Korcsog, 2018-12-23:
@archiolochos did you use a Sonoff?

RavenSystem, 2018-12-25:
@Korcsog Added to [TODO List])https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List)
@archiolochos Reverse relays is planned since some time ago, but it's very strange how your garage door works. As @Korcsog said, do you use a Sonoff device?

Korcsog, 2018-12-25:
@RavenSystem Thanks! Also mention the offset option for humidity too, DHT-s are sometimes inaccurate at humidity measuring compared to other sensors ( I’m using BME280s with esp-HomeKit, DHT shows 20% more humidity). 
@archiolochos If you using an external relay, i think you’re using an active low type relay watch works reverse than others.

RavenSystem, 2018-12-25:
@Korcsog I have added humidity offset to TODO list too.

Korcsog, 2018-12-26:
@RavenSystem  As i've mentioned earlier, the partly working esp-homekit-motor code.
Maybe it's helpful, because I think lot a people waiting that window covering support 😄 

```
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <queue.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include "button.h"

#define DURATION 10

static QueueHandle_t window_queue = NULL;

typedef enum {
    window_command_idle,
    window_command_open,
    window_command_close
} window_command;

typedef enum {
    window_state_idle,
    window_state_opening,
    window_state_closing,
} window_state;

static int duration = 0; //seconds
static window_state local_state = window_state_idle;
static int local_current_position = 0;
static int local_target_position = 0;

void on_update_target_position(homekit_characteristic_t *ch, homekit_value_t value, void *context);
homekit_value_t getter_current_position();
homekit_value_t getter_position_state();
homekit_value_t getter_target_position();
homekit_characteristic_t current_position = HOMEKIT_CHARACTERISTIC_(CURRENT_POSITION, 0, .getter=getter_current_position);
homekit_characteristic_t target_position  = HOMEKIT_CHARACTERISTIC_(TARGET_POSITION, 0, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(on_update_target_position), .getter=getter_target_position);
homekit_characteristic_t position_state   = HOMEKIT_CHARACTERISTIC_(POSITION_STATE, 0, .getter=getter_position_state);

const int led_gpio = 2;
const int up_gpio = 12;
const int down_gpio = 4;
const int button1_gpio = 0; //14=D5 //0=button //9=reset //5=D1
const int button2_gpio = 5; 

#define start_up() gpio_write(up_gpio, 1);
#define stop_up() gpio_write(up_gpio, 0);
#define start_down() gpio_write(down_gpio, 1);
#define stop_down() gpio_write(down_gpio, 0);
#define led_on() gpio_write(led_gpio, 0);
#define led_off() gpio_write(led_gpio, 1);

void gpio_init() {
    gpio_enable(led_gpio, GPIO_OUTPUT);
    gpio_enable(up_gpio, GPIO_OUTPUT);
    gpio_enable(down_gpio, GPIO_OUTPUT);
    gpio_enable(button1_gpio, GPIO_INPUT);
    gpio_enable(button2_gpio, GPIO_INPUT);
    stop_up();
    stop_down();
    led_off();
}
[... 219 lines cut ...]
    uint8_t macaddr[6];
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    
    int name_len = snprintf(NULL, 0, "ESP Homekit Motor %02X:%02X:%02X",
            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "ESP Homekit Motor %02X:%02X:%02X",
            macaddr[3], macaddr[4], macaddr[5]);
    
    name.value = HOMEKIT_STRING(name_value);
}

void on_wifi_ready() {
    homekit_server_init(&config);
    window_init(DURATION);
    buttons_init();
}

void user_init(void) {
    uart_set_baud(0, 115200);
    create_accessory_name();
    gpio_init();
    wifi_config_init("ESP-Homekit-Motor", NULL, on_wifi_ready);
}
```

Korcsog, 2018-12-27:
@RavenSystem Another remark when thermostat accessory is selected:
is it possible to read the temperature/TH sensor first at startup and then set its reading to the defined interval between sensor reads? My problem is that the sensor shows 0.0 degrees until the first sensor reading happens and it can be quite disturbing at first start when you set up 30-60 sec reading interval.

closed 2018-12-31 by Korcsog