RavenSystem/esp-homekit-devices #144: GPIO14 - Button Mode

homkittoby, 2018-12-24:
Problem with Button Mode:
If someone presses too long, the configuration will be reset.

Idea new Option for GPIO14 -> Only Button Mode

homkittoby, 2018-12-25:
@RavenSystem 
If you use the function 9. Socket + Button + TH sensor. 
Example push button 1x light on, push button 2x TV on. 
If your child now plays the button and presses 10sec, your configuraton is deleted. 
If your child now plays the button and presses for 3sec, your relay turns on. 
Unfortunately happened several times. That's why it's a bug for me.
A port with only the button function would be great. 
So came the idea to make the GPIO 14 function configurable. 
1. Standard, 2. Toggle and new 3. Button function.

seritos, 2018-12-26:
@homkittobt

Maybe an option would be to setup the time for a reset. I also have kids and they have performed a reset accidentaly.

What I would like to purpose is the possibity of enable the button (GPIO 0) to be used in switch mode( so it could be easily integrable with our home installations ) and then use the GPIO 2 for TH/Other.
Then we could maximice the potential of the board.
If there would be aditional GPIOS in the board, then use them for the TH, and leave the GPIO 2 for switch mode. However, i have readen that the GPIO 2 has no pull up resistor.

Is this an option @RavenSystem? 

I take advantage of the opportunity to thank you for such an excellent implementation.

RavenSystem, 2018-12-26:
Ok, final solution: a factory default reset will can be done only for the first 2 minutes since device boots. 

@seritos gpio2 is the replacement for gpio14 in the new board layout. Device type 1, that can be used with Sonoff basic, s20 and T1 ch can use the gpio14/goio2 to connect an external toggle switch (not button). But if you want a sonoff basic with externar switch and a th, for example, better pay $5 and buy an additonal sonoff and use it with a th sensor. The esp8266 is a chip that has low resources, and HomeKit needs many resources due to strong cypher and Bonjour. Is it possible to manage all with an unique Sonoff? yes, but it is not recommendable, and for $5 extra, well, I think nobody has ruined for that. It is similar to use 4 water valves with a Sonoff 4ch: a single esp8266 managing 4 timers and notifying all of them to HomeKit ecosystem every second; better use 4 Sonoff basic.

Of course, thanks for your feedback and suggestions.

seritos, 2018-12-26:
Smart solution to allow the reset only in the first 120 secs!!

Regarding the posibility of Swich + TH, i am fully agree with your explanation. The only argument to do so is the space; it is quite hard to fit the board into the hole, so place two is almost impossible in most of the locations. For sure, this devices are more that affordable for anyone around! :) 

... i have just received the dual one, hope you release smthg soon for the blinds/window rolling. Lots of people are willing to see it! 

Thanx a lot, 

Thanx!

homkittoby, 2018-12-27:
@RavenSystem 

„Ok, final solution: a factory default reset will can be done only for the first 2 minutes since device boots.“

great idea 👌

Thank you very much

closed 2019-01-27 by RavenSystem