RavenSystem/esp-homekit-devices #151: Bug: Garage door opening/closing behavior when stopped

Korcsog, 2018-12-31:
Hi!

I discovered a new bug! I'm installed yesterday the garage door with two sensors. When you open/close and obstruction detected or you stopped the garage door with the remote ( stays in closing/opening state) and you press the button in Home app, the relay pulls twice. So the garage door opens/closes again and the second relay pull stops the door!

Also I've created this for the Wiki page (for lazy peoples 😄 ):

https://i.imgur.com/C2FdT39.jpg

dvelardi, 2018-12-31:
Hi @kristof,
Very interesting your experience!

I’m also would control garage door using homekit assistant: what you used? Shelly or sonoff basic? You modified it? You added some bridge? How you solved problem of relais? You have some schema of your project?

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

https://i.imgur.com/C2FdT39.jpg

Korcsog, 2018-12-31:
@dvelardi Read the [Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki) first! 
It's a "native" HomeKit project, no bridge required. I'm using a Wemos D1 mini, but Shelly and Sonoff also works (this code is for Sonoffs mostly).

RavenSystem, 2018-12-31:
@Korcsog Many thanks! I added pairing code image to the wiki.

Garage Door bug: well, it is not a real bug. The real problem is that it is not possible to know when door is moving. With sensor close and sensor open, you know if door is fully open or fully close, but what I assume is when door is not closed and is not opened, it is moving: if last door state was closed, then door is opening, and if last state was opened, then it is closing. But there is not sensor that can confirm that door is moving.

So, if door is stuck in a half way position (using your remote, for example), HomeKit doesn't know it and thinks that door keeps moving. For a perfect functionally, it's necessary another sensor with a gyroscope and accelerometer to know the real state of the door, or connect device directly to the garage door controller and read its feedback. But both options are complex. With this software you can manage a garage door with HomeKit using a cheap esp8266 device and a couple of magnetic sensors (cheap too).

I am working on implement an obstruction detected function using GPIO5, that could mitigate some lacks of functions; but only mitigate.

Korcsog, 2018-12-31:
@RavenSystem Thanks for the fast answer.
When i've used mqtt based Wemos for garage doors and the door stucked in closing/opening state, the button press in the home app (with one relay pull) stopped the door or started it again. My problem is with the double relay pull at the moment. So the question is, could you modify the code (or add an option) for this behavior so that when the button is pressed in the closing/opening state the relay is pulled only once, not twice?

dvelardi, 2018-12-31:
@kristof I’ve already read wiki...
My doubt is referred to use of two sonoff: I have to use them in parallel? How I have to connect them? In my opinion it’s sufficient use only one sonoff basic with track cutter...what do you think?

DV

digiservice2001, 2018-12-31:
@dvelardi Oh man, U need to learn more about electricity...
but finally: be very careful with electric shock, Sonoff has powered output!!!

dvelardi, 2018-12-31:
I’ve read this on wiki:

“You must use a Sonoff SV with isolated relay. And you can use another Sonoff like Basic or TH with isolated relay (you must do it cutting tracks) and using only a sensor for closed status attached to GPIO14.”

This mean I need only one sonoff?!

DV

RavenSystem, 2018-12-31:
@dvelardi I changed wiki description. Maybe now it is clearer.

i3laze, 2019-01-12:
Seems, we're talking about `Has stop step: True` here.

Actually, we have `Working Time` parameter. It should be handled in this situation correctly:
`If HasOpenedSensor=true and Time > WT and StatusOpened=False`, then Garage Status should become stopped and any HomeKit actions should produce 1 press, as topic starter suggests.

IMHO, this behavior should be regardless having obstruction sensor.

Korcsog, 2019-01-15:
@i3laze Exactly what i've want! When obstruction detected (or the garage door stopped in between fully opened or fully closed position) only 1 relay pull is needed because most garage doors stops automatically. At the moment when the garage door stops between the fully opened/closed position the press on the button (in Home app) performs two relay pulls: first starts the door again, the second stops it and it still stuck in between the fully opened/closed position.

closed 2019-01-27 by RavenSystem