RavenSystem/esp-homekit-devices #152: Ice melt thermostat

i3laze, 2018-12-31:
Ice melt system is a relay with two thermostat controllers to limit working temp:
https://www.midselec.co.uk/erc-snow-and-ice-melting-controller-and-sensor.html
Such system usually turn on the heater when outdoor temp is between -3C and +5C to help icicles melt.

A. One Sonoff with DH22/DS18B20 might be represented as two dependant thermostats (a heater and a cooler). 
Or 
B. A regular heater thermostat with an option (hidden in Eve.app) to turn off above T=5C.

How about such device mode?

RavenSystem, 2018-12-31:
It is not possible. Target temperature of an Apple HomeKit thermostat is between 10ºC and 38ºC.

Korcsog, 2018-12-31:
@i3laze No need for that. I'm using one Sonoff Basic with DS18B20 for this purpose... You could create two automations in the Eve app... one for turning on the thermostat when temp is below e.g. -3 degree and an another one for turning off the thermostat when +5 degree reached!

i3laze, 2018-12-31:
Thanks! Will go Tasmota here.

Happy New Year!

closed 2018-12-31 by i3laze