RavenSystem/esp-homekit-devices #153: Enhancement request: MQTT support

peros550, 2018-12-31:
Hi Jose, 

With your firmware we have achieved to bring into Apple's Home app our devices and control them. Moreover, we have started to minimize the need for middle man devices (like i.e.  a raspberry running homebridge)

With Home App we can do automations but only to the extend that Apple allows. If we want to do more powerful things, we would need to pass Sonoff status to external servers. 

One  common protocol for this job I think is MQTT. 

It would be of great help if Ravencore was able to be customized and send our device's status into an MQTT server. We could implement so many custom ideas based on that functionality. Of course, the opposite would also be welcome (Ravecore subscribes to MQTT and responds to specific messages). The possibilities would be endless. 

Thanks for reading so far. I am not sure how much of work and efforts this idea translates to, I hope it is feasible. 

Thanks & Best Regards

RavenSystem, 2019-01-04:
You can do many others automations that Home App doesn't allow using a third-party software. I use this App:
Home 3: https://itunes.apple.com/us/app/home-3/id995994352

Respect MQTT support, I must say no to this request. Reasons:
- MQTT is insecure.
- Integrate a new framework is impossible because flash size is limited, and firmware size ir near to the limit because OTA firmware use half of flash storage, and RavenCore the other half.
- There is a lot of tasks to do before.

peros550, 2019-01-05:
Hi Jose, 

I understand your reasons. 
I hope try to include MQTT client in custom builds. 

Thanks

closed 2019-01-05 by peros550

RavenSystem, 2019-01-06:
I will stay tuned of your work.