RavenSystem/esp-homekit-devices #154: SONOFF DUAL pilot wire for electrical radiator management

f1jsh, 2019-01-01:
Hello José,
thank you for your development, very nice work! really good!
Consider developing a mode of operation for the management of electric radiator heating modes with pilot wire.
It is very simple to add two diodes in the SONOFF DUAL module to pilot the radiator.
What do you think ?
Thank you and happy new year 2019
Oliver

https://www.planete-domotique.com/blog/2012/01/05/piloter-un-radiateur-grace-a-son-fil-pilote/

[image]

shingle-smith, 2019-01-18:
Without adding diodes to the Sonoff, do you think it would still be possible to trigger 'eco' mode with 240V AC from the relay?

shingle-smith, 2019-01-21:
Thanks - but in theory a signal from a Sonoff Basic single relay through the pilot wire should work as a trigger for eco/réduit mode, correct? (230V pleine alternance)

shingle-smith, 2019-01-21:
I’ve now tested this using a Dimplex Q-Rad electric radiator, which can use a pilot wire energised at 230/240V AC.

When the Sonoff basic is switched off, the heater is in Comfort mode (and starts heating up), and when the relay is on, the heater turns to Setback mode, 5c lower than comfort.

There is no way of setting fully ‘off’ as that would require a diode to filter positive/negative.

f1jsh, 2019-01-21:
The ideal is to take a SONOFF DUAL module and integrate the two diodes.
That way you have all the operating cases.
And that makes a chaffage comamnde very practical and powerful. I asked JOSE to integrate this feature but so far no response.
He must be very busy!

[image]

closed 2019-02-28 by RavenSystem

topzoe, 2019-12-14:
Do you have a HAA Json for SONOFF DUAL for drive pilot wire?