RavenSystem/esp-homekit-devices #155: Dimmer

arturodiazad1, 2019-01-01:
Hello, is it possible to include a dimmer switch type Ewelink Wifi Dimmer in the list of future versions? https://s.click.aliexpress.com/e/Z86Yytb
Or if you know of any device that can be compatible and can control the brightness to 220v, I appreciate it. regards

Kristian8606, 2019-01-03:
I do not have the same dimmer, but if you give more information about the device I will try to modify my code.

RavenSystem, 2019-01-03:
Added to TO-DO list https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

arturodiazad1, 2019-01-05:
Sorry kristian8606 for not answering before, I still do not own the device, I am considering buy it since I have the need to create some dimmer lights in my home, as I read this device has the esp8266 chip, I have been reading in other forums and even the device of the following link that is similar https://es.aliexpress.com/item/Interruptor-de-atenuaci-n-LED-WiFi-220-V-110-V-interruptor-de-Panel-de-atenuaci-n/ 32859257784.html use it with the tasmota software. Thank you very much, I have a lot of enthusiasm for this project.

LyoshaDovolen, 2019-02-03:
https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer#multiple-manufacturers-touch-eu-and-us