RavenSystem/esp-homekit-devices #158: Sonoff Dual R2 AC interference

sblitzer, 2019-01-03:
Hi,

I recently installed Sonoff Dual R2 - and successfully flash it with Ravencore rev 0.5.6 and put in OTA setting.
HomeKit I successfully connecting and switching the 2 relays BUT , every time relay is switch ON or OFF it's all other ravencore devices in my house are disconnecting.
I have 15 sonoff basic flashed with 0.5.6 in my house that work perfectly with homeKit.
P.S  - I use the sonoff dual for opening and closing my Blinds.

Can you  please assist with this phenomenon

Thanks

sblitzer, 2019-01-03:
I made scenario in the Home App (IPhone) 
scenario OPEN Blinds : Relay1 on Relay2 off
scenario close Blinds : Relay1 off Relay2 on

peros550, 2019-01-03:
> I made scenario in the Home App (IPhone)
> scenario OPEN Blinds : Relay1 on Relay2 off
> scenario close Blinds : Relay1 off Relay2 on

Friendly speaking, I believe you need something better than this approach to make 100% sure that Relay1 and Relay 2 will never close circuit at the same time. Perhaps an electrical design modification could provide this guarantee. 

In case of any potential failure, closing two relays at the same time will most probably make your blind/windows motors fail permanently.  

In regards to the "no response" issue you reported, I have been experiencing random "no response" status as well. Somehow, I also suspect that one device may interfere to all others. 

I don't have evidence of this speculation, but it's quite interesting that you also report similar behavior.

sblitzer, 2019-01-03:
My blinds driver protect the motor if there is such case of failure I have verified it with the manufacturer. So I didn’t think any failure will be possible.

For the AC interference with the RF / WiFi this is probably code bug (I hope 😝)

RavenSystem, 2019-01-03:
Automations are not good to do what you want because there is a moment that both relays are connected and this will generate an AC interference produced by the protection of your motor driver. I suggest you to try this other setup:

scenario OPEN: Relay1 off (and nothing more)
automation: when relay1 go off, relay2 go on.
scenario CLOSE: Relay2 off (and nothing more)
automation: when relay2 go off, relay1 go on.

Or better wait until I release blinds support for Sonoff Dual.

sblitzer, 2019-01-03:
Thank you for your reply, I think I’ll wait for your release 
And thank you for all your work 😎

seritos, 2019-01-09:
@RavenSystem any ideas of when will you release the version for the blinds? 

Thanx!

RavenSystem, 2019-01-27:
> @RavenSystem any ideas of when will you release the version for the blinds?

No idea, sorry.

closed 2019-01-27 by RavenSystem