RavenSystem/esp-homekit-devices #160: Bridge

Kristian8606, 2019-01-04:
how to build a bridge on esp if is  possible?

Kristian8606, 2019-01-04:
my idea is to use sensors for battery and for that I need a bridge

maccoylton, 2019-01-04:
I'm working (slowly) on bringing an old bluetooth version of homekit than runs on NRF51822 up to date, for the purpose of creating sensor that can be battery powered. The current version can be compiled and run, but only support limited set of accessory type, it also has to be paired via an IOS11 device. You can see it here:- https://github.com/maccoylton/NRF51HomeKitBLE

peros550, 2019-01-04:
I find your idea very interesting. 

Assuming you want to build a remote temp/humidity sensor, I believe the code in the hub would have  common parts with the thermostat example. 

A major difference would be how the temperature values are read. Instead of reading a physically connected sensor, you would need to receive those values wirelessly and store them. In this way, you can have an ESP board constantly connected to wifi & power (that would be the hub), and multiple esp boards acting as remote sensors which send measured values wirelessly to the hub and then switch into power saving mode. 

Eventually, you would need to pick a connection type & protocol between the remote sensor and the hub and create two firmwares, one for the sensor and one for the hub. 

That's how I see it.

Kristian8606, 2019-01-04:
Тhat's the idea.
I noticed this homekit_accessory_category_bridge, and I thought it might be possible to build it

RavenSystem, 2019-01-05:
@peros550 A bridge HomeKit <-> MQTT could be a beautiful project.

peros550, 2019-01-05:
> @peros550 A bridge HomeKit <-> MQTT could be a beautiful project. (#153)

It's definitely worth the effort. I'll put it in my to dos. I'll let you know with any developments or news as soon as they become available.

closed 2019-01-27 by RavenSystem