RavenSystem/esp-homekit-devices #161: IMPORTANT NOTICE!! RavenCore 0.5.7 will remove all configured user settings

RavenSystem, 2019-01-04:
RavenCore version 0.5.7 will be released soon, and some changes has been done. One of them is the recode of all storage system, that affects to saved user settings. All saved settings will be removed.

**It is highly recommended to erase flash and install it again from scratch.**

I understand that this is annoying for many users (for me too), I apologize but it is necessary.

RickNunn, 2019-01-04:
Will this include saved wifi settings?

RavenSystem, 2019-01-04:
> Will this include saved wifi settings?

No, but I repeat **"It is highly recommended to erase flash and install it again from scratch."**

RavenSystem, 2019-01-05:
RavenCore 0.5.7 released.

closed 2019-01-05 by RavenSystem

HomeACcessoryKid, 2019-01-06:
I propose to reopen this issue and discuss a very simple remedy for this situation.

We are talking here about the size of the labels in sysparam memory which indeed is wasteful.

You are replacing e.g. "show_setup" with "a" etc...

Now, it is not needed to re-flash the device in case you add  the following to the beginning of user_init:
```
sysparam_get_bool("show_setup",temp);
if (SYSPARAM_OK) {
    sysparam_set_string("show_setup",""); //like this the old param will be deleted
    sysparam_set_bool("a",temp);
}
//and all other 26 params as well
```
In a year from now you can remove this code if you want, but for now I really think people will appreciate not having to configure again and certainly not having to erase flash.
What is more, I think that in the current situation most people will reconfigure using EVE and not erase flash at all.
But that means those old parameters will stay around as active and the flash sector will wear out that much faster.

Like I said, this is the beginning of a dialogue, so please share your view?

TIA,
HacK

HomeACcessoryKid, 2019-01-06:
Thanks for fixing this in 0.5.8 !!!!!