RavenSystem/esp-homekit-devices #162: Feature Request — Device Type Fan & Light

RickNunn, 2019-01-04:
Would it be possible to have the option to declare the device type to HomeKit as a Fan, and also as a Light? 

I know it's sill just on/off functionality but it would be nice. It would also give you the ability to select other Light icons in the Home app.

I am aware that you can set the "Type" to Light, and Fan in the Home app.

digiservice2001, 2019-01-04:
U can do it yourself in the Home app...👍🙄

AramVartanyan, 2019-01-04:
It’s that easy:
Home app -> Force touch on the “Switch” accessory -> Settings -> tap on “Type” -> select Light, Fan or Switch.

RavenSystem, 2019-01-04:
@RickNunn Please, don't leave empty issue description.

> Would it be possible to have the option to declare the device type to HomeKit as a Fan, and also as a Light?
>
> I know it's sill just on/off functionality but it would be nice. It would also give you [...]

Seems that nobody read "I am aware that you can set the "Type" to Light, and Fan in the Home app."

I know what you want. You want to declare your device as a light bulb instead a switch (for example). Well, a lightbulb can be configured with same characteristics of switch, but a fan has many different characteristics compared with a switch accessory type, and doing that requires many additional code and flash storage.

But as @AramVartanyan says, you can select light, and then you are able to select a different icon for your accessory. Those icons are same than a lightbulb.

closed 2019-01-27 by RavenSystem