RavenSystem/esp-homekit-devices #163: Some requests and possible bugs.

seritos, 2019-01-04:
Hi Jose Antonio, 

I have some suggestions for you and some issues that I believe they are bugs:

Would be possible to have a "intercom" mode? This option will enable an output for 5/8 ( maybe configurable from the EVE app) seconds. Enough time to open the door. 

For the Sonoff Dual and the future blind control: The switches for the blinds many times stands in one position enabled, I mean: If you want lift it up completely,. you press the up button and leave the room. the blind will finish its jobs and the roller motor will stop at the end position, but remember that the switch remains ON. this is a completely safe mode to work in manual, as I said, the roller engine stops when finishes. I just wanted to explain this to keep it in mind when programming the desired blind control for Sonoff Dual.

Possible BUG: The S20 has no GPIO2/14. How can a thermometer sensor be installed? I have tried all possibilities and can't make it work. This would be very interesting option to be used in the S20. I installed tasmota to make sure that the sensor and soldering was ok, and worked perfectly. 

This last point has been previously exposed, and you have shown you disagree; The possibility of using in a basic Sonoff in switch mode with the GPIO0 and left the GPIO2/14 for sensors. I am fully agree with you that they are very cheap and perfectly affordable, however, there problem is the space to integrate the modules. in some cases, there is only space for one, and the possibility of enable the sensor would be awesome. 

Once again, thanx a lot for your effort and dedication.

RavenSystem, 2019-01-04:
> Would be possible to have a "intercom" mode? This option will enable an output for 5/8 ( maybe configurable from the EVE app) seconds. Enough time to open the door.

No, until Apple release an accessory that match.

> Possibility of see the temperature with 0,1 degree resolution. In don't know is the own HomeKit system allows this. I can see it perfectly in the EVE app (in 0,1 degree resolution)

HomeKit allows it, and this is the reason that in other Apps you cat see temperature with a 0.1 degree resolution. Apple rounds it in Home App with a 0.5 resolution.

> Possible BUG: The S20 has no GPIO2/14. How can a thermometer sensor be installed? I have tried all possibilities and can't make it work. This would be very interesting option to be used in the S20. I installed tasmota to make sure that the sensor and soldering was ok, and worked perfectly.

The S20 is not intended to install a sensor easily. S22 is better for that.

> This last point has been previously exposed, and you have shown you disagree; The possibility of using in a basic Sonoff in switch mode with the GPIO0 and left the GPIO2/14 for sensors. I am fully [...]
Moreover, the pad for the GPIO2 is extremely small and I have already broke 2 of them, so there is no way of using them again ( I can't make such a small solder to do it directly on the chip) in switch mode ;-(

And using a Shelly1 and a Sonoff Basic?

seritos, 2019-01-05:
> The S20 is not intended to install a sensor easily. S22 is better for that.

I ask in the other post if this was possible with the S20, and you said it was. You mean that if I would directly solder into the chip pin would work? GPIO2 or GPIO14? then this could be an option 👍 

> And using a Shelly1 and a Sonoff Basic?

I have already thought about this possibility, also in the Sonoff Dual. The main purpose is to control the temperature of multiple places and made automations accordingly. Would it be possible, at least, to launch it as a test/beta mode? 
If works fine, then you could consider the possibility of release it officially. :-)

Thanx

RavenSystem, 2019-01-05:
On S20, you can solder a wire to GPIO2 (board type 2) or GPIO14 (board type 1) and then connect to it a TH sensor (remember that sensor must have built-in pull-up resistor and capacitor). This should work. It is recommended to put a 0 ohm resistor between GPIO and sensor data wire for protection (S22 and Sonoff TH have it).

The other case: as far I know, you want a device that:
a) Switch HomeKit triggered with an external mount-wall toggle switch using GPIO0.
b) A TH sensor connected to GPIO14.
Is it right?

seritos, 2019-01-05:
Exactly. The possibility of choose the switching input for the GPIO0 and leave the GPIO 2/14 for the sensor. 
....would this affect to the reset process if the decides rebooted and the switch is on? 

Regarding the S20, I will try very soon al let it you know. 

By the way, this morning I flashed a T1 ( the touch model ). It works perfectly with RavenCore..  I might buy a dual touch for the blinds instead of the Sonoff Dual. The future blind control version will allow this Sonoff Touch Dual? 

Happy magic kings night to everyone!

seritos, 2019-01-05:
Working flawlessly with DHT22 in S20 in GPIO2. now it can perfectly be used as a thermostat or similar.

[image]

closed 2019-01-27 by RavenSystem