RavenSystem/esp-homekit-devices #164: Wifi Client mode >> Wifi AP mode >> Wifi Client mode

peros550, 2019-01-05:
Hi Jose, 

Correct me if I'm wrong, but I have understood that when an already paired device restarts, it tries to reconnect to the last known Wifi SSID for 180 seconds (DWIFI_CONFIG_CONNECT_TIMEOUT=180000) and then if it does not succeed it turns into AP mode. 

I was wondering if you could create a timeout for the period operating in AP mode. For example it could stay in AP and wait for any connections and then if nothing happens for i.e. 5 minutes it would try to reconnect again to the last known Wifi network. 

In the current implementation if for whatsoever reason the device does not connect to Wifi SSID, it turns into AP mode and stays there until you physically disconnect it from the mains. 

Thanks & Regards

RavenSystem, 2019-01-06:
It's a good idea, but this must be requested in esp-wifi-config repo: https://github.com/maximkulkin/esp-wifi-config

peros550, 2019-01-08:
Thanks Jose! I have just opened this issue: https://github.com/maximkulkin/esp-wifi-config/issues/11

closed 2019-01-08 by peros550

AramVartanyan, 2019-02-04:
Is there someone who is willing to help testing this workaround?

https://github.com/AramVartanyan/esp-wifi-config

I have added a 10 min timeout in AP mode. It is just an idea for now.

peros550, 2019-02-04:
> Is there someone who is willing to help testing this workaround?
> 
> https://github.com/AramVartanyan/esp-wifi-config
> 
> I have added a 10 min timeout in AP mode. It is just an idea for now.

I tried your version. The device does not switch into AP mode after following interval: DWIFI_CONFIG_CONNECT_TIMEOUT=15000

AramVartanyan, 2019-02-04:
> I tried your version. The device does not switch into AP mode after following interval: DWIFI_CONFIG_CONNECT_TIMEOUT=15000
Good. I will try to find a way out. Thanks

peros550, 2019-02-05:
Hi I tested the updated version. It seems to work fine! 

Here are my two test cases: 

a) Turn off the Wifi at the router  AND power reset esp8266 (Wemos) .Operation is as expected meaning AP fires up, then when Wifi becomes available again,  devices gets connected to the known Wifi. 

b) Turn off only the Wifi at the router, leaving the esp8266 device (Wemos) powered on, then  it just  starts trying to reconnect and never fires up the AP.

To be honest, in my previous response I had only tested test case "B". 

Many thanks for your help

AramVartanyan, 2019-02-05:
You’re welcome. This issue was bothering me for months.
Case b) is a normal behavior related to the sdk auto reconnect function.
This workaround will be helpful until the real bug is found.
Since Maxim himself is on it, I believe it will be soon.

Regards

aslater3, 2019-02-09:
I suffer from this issue as well. My home network is virtualised and when recovering from a power outage it can take 5 mins or more for everything to reboot, by which point the SONOFFs running ravencore have given up trying to connect. It would be better if, once configured they stay out of AP mode until some form of user intervention? E.g. a button press for a certain period of time perhaps?