RavenSystem/esp-homekit-devices #165: Enhancement request: support for sonoff ifan02

vmarks, 2019-01-05:
Ifan02 is a ceiling fan controller supporting light on/off and four fan speeds (off, slow, medium, high) over Wi-Fi. I would really like to use it with apple homekit.

RavenSystem, 2019-01-06:
Ok. If you can, please, put GPIO map of this device.

vmarks, 2019-01-06:
Ok!
(How do I do that? It’s supported by tasmota firmware, and I have been able
to flash it.)

Partizan7676, 2019-01-09:
Try find info in Tasmota repo. Looks like what you ask:
  { "Sonoff iFan02",   // Sonoff iFan02 (ESP8285)
     GPIO_KEY1,        // GPIO00 WIFI_KEY0 Virtual button 1 as feedback from RC
     GPIO_USER,        // GPIO01 ESP_TXD Serial RXD and Optional sensor
     0,                // GPIO02 ESP_LOG
     GPIO_USER,        // GPIO03 ESP_RXD Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 WIFI_O2 Relay 3 (0 = Off, 1 = On) controlling the fan
     GPIO_REL2,        // GPIO05 WIFI_O1 Relay 2 (0 = Off, 1 = On) controlling the fan
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     GPIO_KEY2,        // GPIO09 WIFI_KEY1 Virtual button 2 as feedback from RC
     GPIO_KEY3,        // GPIO10 WIFI_KEY2 Virtual button 3 as feedback from RC
                       // GPIO11 (SD_CMD   Flash)
     GPIO_REL1,        // GPIO12 WIFI_O0 Relay 1 (0 = Off, 1 = On) controlling the light
     GPIO_LED1_INV,    // GPIO13 WIFI_CHK Blue Led on PCA (0 = On, 1 = Off) - Link and Power status
     GPIO_KEY4,        // GPIO14 WIFI_KEY3 Virtual button 4 as feedback from RC
     GPIO_REL4,        // GPIO15 WIFI_O3 Relay 4 (0 = Off, 1 = On) controlling the fan
     0, 0
  },

Partizan7676, 2019-01-10:
@RavenSystem was it helpful or not?

RavenSystem, 2019-01-11:
@pandarek Yes! But I need the way ifan02 relays work and manage fan.

angyp, 2019-01-12:
GPIO_REL2, // GPIO05 WIFI_O1 Relay 2 (0 = Off, 1 = On) controlling the fan = Low 

GPIO_REL3, // GPIO04 WIFI_O2 Relay 3 (0 = Off, 1 = On) controlling the fan + GPIO_REL2, // GPIO05 WIFI_O1 Relay 2 (0 = Off, 1 = On) controlling the fan = Medium

GPIO_REL4, // GPIO15 WIFI_O3 Relay 4 (0 = Off, 1 = On) controlling the fan = High

To prevent motor burnout from starting on low or medium, I think the high relay should be triggered for about 3 seconds at the start.

RavenSystem, 2019-01-13:
Is it right?

Off -> Relay 2: OFF; Relay 3: OFF; Relay 4: OFF
Low -> Relay 2: ON; Relay 3: OFF; Relay 4: OFF
Med -> Relay 2: ON; Relay 3: ON; Relay 4: OFF
High -> Relay 2: OFF; Relay 3: OFF; Relay 4: ON

Starter -> Always set to High for 3 seconds.

RavenSystem, 2019-01-13:
@angyp Excellent!!!

seritos, 2019-01-13:
Just FYI. The iFan02 is at half price in Aliexpress...don't know how much will it last.

Partizan7676, 2019-01-13:
also interesting for that device to use it for kitchen hood

vmarks, 2019-01-18:
What else is needed? I'm ready to test firmware as soon as it is available!

RavenSystem, 2019-01-27:
This feature will be implemented, but later.

vmarks, 2019-05-26:
I can't wait! I'm excited for this.

vmarks, 2019-06-27:
Hi, 

Is there anything to test? I am still very interested in this.

vmarks, 2019-07-19:
Is this feature implemented?

Can I test?

RavenSystem, 2019-07-20:
Sorry, it is not implemented yet.

vmarks, 2019-07-20:
Thank you - I am very hopeful it will be implemented someday soon.

I would offer to send you the hardware, but it's 125vac, and Europe is
240vac. I don't know if that would be helpful.

RavenSystem, 2019-07-22:
> I would offer to send you the hardware, but it's 125vac, and Europe is
240vac. I don't know if that would be helpful.

Don't worry. I think I have all info needed, but none of actual development uses a fan type accessory. Fans will be implemented with HAA, but not soon.

toshibochan, 2020-01-12:
Hope soon see sonoff iFan working in HomeKit!

RavenSystem, 2020-02-21:
@vmarks Install HAA v1.7.0 and use this:

HAA JSON config for Sonoff iFan02:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"0":{"r":[{"g":15},{"g":4},{"g":5}]},"y0":[{"v":1,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":4},{"g":5,"i":0.6}]}},{"v":35,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":4,"i":0.6},{"g":5,"i":0.6}]}},{"v":70,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":5}]}}],"b":[{"g":9}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}`

I hope it works. Please, leave feedback.

Partizan7676, 2020-02-22:
waiting for test? @vmarks good luck

RavenSystem, 2020-02-22:
@Partizan7676 I know, but that JSON is for 02, not for 03. Have you a iFan03?

Partizan7676, 2020-02-22:
@RavenSystem i dont have it, but plan to buy.

RavenSystem, 2020-02-25:
@vmarks Did you test it?

egorist, 2020-02-26:
I tried it with iFan03, it works only with the fastest speed. and no light (relays are mixed as I see. If anyone make json I can try it...

egorist, 2020-02-26:
I made it for iFan03 myself. I don't know about light (relay is broken), but everything alse (fan control) works good. 

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":1,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":35,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":70,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":14}]}}],"b":[{"g":9}]},{"0":{"r":[{"g":9}]},"1":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]}

closed 2020-02-27 by RavenSystem

brymonster, 2020-03-05:
@egorist 
Does the iFan03 lose the RF remote functionality when flashed?

Partizan7676, 2020-03-06:
> @egorist
> Does the iFan03 lose the RF remote functionality when flashed?

good question! also interest for that. @egorist  , is RF remote working too?

tonysprenk, 2020-03-06:
Probably it wil still work but the status will not be updated in the home app. Example: the fan is off in the home app, it is turned on with the RF remote, the fan will be on physically but the home app will say that it is off.

seritos, 2020-03-06:
The remote functionality is lost in the Sonoff RF. I would say you lost this functionality on the iFan, but if its not handled by the esp8266, then you might be able to keep it. RavenSystem will tell you for sure. You can also ask it in the official forum

pilot1981, 2020-03-06:
So, your ESP8266 on IFan03 don't receive signal on GPIOxxx from RF module? I'm interested because I have to do a difference project using Sonoff basic RF R3 and Sonoff DW1...any info could be useful to everyone!

egorist, 2020-03-06:
> @egorist
> Does the iFan03 lose the RF remote functionality when flashed?

sorry, but I don't use RF

Jeroen207, 2020-04-09:
Is there someone who build a pre flashed version for the ifan?

pilot1981, 2020-04-09:
@egorist can you do some test with RF for us, please?

Jeroen207, 2020-04-09:
@egorist, oh yes. If you could implement the RF. I would even pay you for beer.

felsmann, 2020-04-27:
Great stuff over here.  So, noob here I want to get iFan03 working with homekit, what’s the simplest guide I can follow?

vmarks, 2020-07-06:
> waiting for test? @vmarks good luck
> 

Following up (sorry to revive this, I was sick for a while and just got it done today)

Using HAA 2.5.0 on iFan02 light and fan function.

I noticed from 1.8.5, there was this note:Fan: Added user setting to customize speed steps with "e" key. Default is "e":100.

I do not have steps, I have % all over the place, not sure how to use steps. 

I also notice that iFan02 beeps twice seems to go its fastest at 52%. at 29% it beeps once, and slows down. On original iFan firmware, it beeps once for slow, twice for medium, three times for fast. I'm not finding that 3x beep setting yet. 

at 80% to 100%, the fan slows to a stop. At 52% to 56% ish, it speeds back up. 

My JSON is 

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"0":{"r":[{"g":15},{"g":4},{"g":5}]},"y0":[{"v":1,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":4},{"g":5,"i":0.6}]}},{"v":35,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":4,"i":0.6},{"g":5,"i":0.6}]}},{"v":70,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":5}]}}],"b":[{"g":9}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}`