RavenSystem/esp-homekit-devices #166: Shelly1

Codi878, 2019-01-05:
Hi Jose,

 I am tested the module on 5.7 firmware -  device12 and unfortunately shelly works very unstable. I tested on 230v. 
The module after power on is available and after some time the lottery is visible and then it is not responding in the home app.
On firmware Kelly Tron is the same problem.

RavenSystem, 2019-01-06:
KellyTron and 0.5.7 are same code for Shelly1. Forgot KellyTron.

Try RavenCore 0.5.8 and set external toggle to 0.

GPL71, 2019-01-07:
@RavenSystem hello 
I have a question about Shelly
Shelly is made up because I can use the toggle on wall mount in the room and at the same time the WiFi connection. 
The Raven firmware allow same thing? 
And in Home App the state of wall mounting toggle is updated and managed?

Thank you for your work .... At the moment the best I have seen

GPL71, 2019-01-07:
Hello @RavenSystem

Please help me understanding better Shelly firmware

Original firmware allow me using mount-wall switch and WiFi switch by the App.
Is the same using RavenSystem firmware?
Of course I'll connect the originally mount-wall of the room to the Shelly.
The question is if the state of mount-wall switch will be updated on HOME APP and managed.
It would mean if I switch on mount-wall switch, the icon in HOME APP will turn ON the state.
And then I'll be able to switch off the device
If so it's wooooooonderfull!!!!!

Thank you for your costantly great work

RavenSystem, 2019-01-07:
@GPL71 yes, you must set to 2 External Toggle option in EVE.

GPL71, 2019-01-07:
Wooooooooooowwwwwwww!!!!!!!!!!!!!!!!
Fantastic!!!!!!!
Then it mean I'll switch on the mount-wall switch and Home App will show ON....
And then I'll tap the icon on Home App and It will turn OFF.

Another question .... Consider the follow situation
Mount-wall switch ON and light turn ON ... Home App Icon state turn ON
Now .... Tap Icon in Home App and Shelly turn OFF, but mount-wall switch is ON again.
What happen now? 
Does light turn ON again caused mount-wall switch?
Or it stay in the last state (OFF from App)?
It would be fantastic if light will turn ON when I'll press again the mount-switch

I hope it's clear

GPL71, 2019-01-07:
@Ravensystem

Wooooooooooowwwwwwww!!!!!!!!!!!!!!!!
Fantastic!!!!!!!
Then it mean I'll switch on the mount-wall switch and Home App will show ON....
And then I'll tap the icon on Home App and It will turn OFF.

Another question .... Consider the follow situation
Mount-wall switch ON and light turn ON ... Home App Icon state turn ON
Now .... Tap Icon in Home App and Shelly turn OFF, but mount-wall switch is ON again.
What happen now?
Does light turn ON again caused mount-wall switch?
Or it stay in the last state (OFF from App)?
It would be fantastic if light will turn ON when I'll press again the mount-switch

I hope it's clear

RavenSystem, 2019-01-07:
> Then it mean I'll switch on the mount-wall switch and Home App will show ON....
> And then I'll tap the icon on Home App and It will turn OFF.

Yes.

> Another question .... Consider the follow situation
> Mount-wall switch ON and light turn ON ... Home App Icon state turn ON
> Now .... Tap Icon in Home App and Shelly turn OFF, but mount-wall switch is ON again.
> What happen now?

Light turn OFF.

> It would be fantastic if light will turn ON when I'll press again the mount-switch

Light will turn ON when you press again the mount-wall switch although mount-wall switch will be off.

Forget mount-wall switch mechanical status. When you toggle/press/change/trigger mount-wall switch, Home App Icon state and light change.

GPL71, 2019-01-07:
Wooooooooooowwwwww ..... I think u'r a genius !!!!!  That's incredible!!!!
Great work!!! Really!!

In my opinion only one type of device is missing at moment for Shelly and Sonoff
I don't know the correct word in english for the function... 
Is the temporary or momentary closing of relay (1 second). 
It's similar the garage door, where the relay is energized for a time (1 second in this case) without the other security controls.
It would be perfect for the Sonoff SV provided of dry contact, Basic and Shelly boards 
What's the name? Inching relay? 
This further function would complete the programmable devices

I'd just purchased 4 Shelly to test them with you firmware... :D :D
Thank you again for your great work

GPL71, 2019-01-07:
Wooooooooooowwwwww ..... I think u'r a genius !!!!! That's incredible!!!!
Great work!!! Really!! @RavenSystem

In my opinion only one type of device is missing at moment for Shelly and Sonoff
I don't know the correct word in english for the function...
Is the temporary or momentary closing of relay (1 second).
It's similar the garage door, where the relay is energized for a time (1 second in this case) without the other security controls.
It would be perfect for the Sonoff SV provided of dry contact, Basic and Shelly boards
What's the name? Inching relay?
This further function would complete the programmable devices

I'd just purchased 4 Shelly to test them with you firmware... :D :D
Thank you again for your great work

dvelardi, 2019-01-07:
> Risposta: RavenSystem/esp-homekit-devices <reply@reply.github.com>
> Data: lunedì 7 gennaio 2019 15:03

Also me interested in this feature!

I would use sonoff SV to activate relay for 1 second to open garage door: it’s possible?

daniele velardi

GPL71, 2019-01-07:
@dvelardi .... se usi sonoff SV imposti il device 8 e regoli i tempi di apertura del cancello. poi se vuoi imposti anche i finecorsa

@RavenSystem about momentary relay ... In my mind I image a code sequence similar the garage door function..
- push icon or say the command
- relay ON
- Software timer for 1 second
- relay OFF
- reset of Home App icon

The vocal command will be the same ... "hey siri turn on the light" but the relay will be on only for a second
These  feature is suitable for italian electrolock too. "hey siri open door" and relay stay up for a second

RavenSystem, 2019-01-07:
@GPL71 You can do that with an automation. "When switch turn on -> switch turn off".

dvelardi, 2019-01-07:
@GPL71 a me interessa solo il segnale di apertura, poi il cancello si chiude da solo, quindi uso firmware ravencore con device type 8 su SONOFF SV?

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

GPL71, 2019-01-07:
@GPL71 You can do that with an automation. "When switch turn on -> switch turn off".
@RavenSystem ok this is an idea .... but how to?
Home App doesn't allow to do it.... Then which app?

GPL71, 2019-01-07:
@GPL71 a me interessa solo il segnale di apertura, poi il cancello si chiude da solo, quindi uso firmware ravencore con device type 8 su SONOFF SV?
@dvelardi Si .... tu imposti il tempo dello switch ad 1 secondo e non usi i sensori nella configurazione. Collega il relè dell' sv al selettore a chiave
Yes .... use device type 8. set the time to 1 sec without using configuration of sensors or switches. connect the SV relay to the key switch terminal. The board of gate will manage the automation

dvelardi, 2019-01-08:
> @GPL71<https://hes32-ctp.trendmicro.com:443/wis/clicktime/v1/query?url=https%3a%2f%2fgithub.com%2fGPL71&umid=5955090a-d922-4078-b683-f0a19fb00054&auth=48278bb8638c59b19e6a5c9ee0648ca7e9c732c1-de7d9cbc7e709b702789155dd7dc6b30b5126dd1> You must create an ambient and include the accessory in off state. Then, create an automation "When switch on -> exec ambient".

@GPL71 che contatto posso usare per il contatto magnetico di chiusura porta?

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

GPL71, 2019-01-08:
che contatto posso usare per il contatto magnetico di chiusura porta? @dveladi usa il finecorsa di cancello chiuso o aperto o aggiungi un microswitch o un contatto magnetico sul cancello in posizione di chiuso
prova a postare schema cancello che vediamo

you can use the microswitch for closed or opened position or even add a microswitch o magnetic contact on the frame of gate in the position of gate complety closed.
Try to post the diagram of your gate and we'll try find a solution

GPL71, 2019-01-08:
@GPL71 I have one Shelly1 and works like a charm, mounted into an electrical box to control the light of the entrance of my house. I can use the external mount-wall switch, an iOS device, or even automations; and all of them works. But @Codi878 has a lot of problems with his Shelly1 (AC interferences).

@RavenSystems ok ... in couple of 2 weeks I'll have the shelly1 and I'll be able to test them and I'll be available to test for you
Thanks

GPL71, 2019-01-08:
che contatto posso usare per il contatto magnetico di chiusura porta? 
@dvelardi usa il finecorsa di cancello chiuso o aperto o aggiungi un microswitch o un contatto magnetico sul cancello in posizione di chiuso
prova a postare schema cancello che vediamo

you can use the microswitch for closed or opened position or even add a microswitch o magnetic contact on the frame of gate in the position of gate complety closed.
Try to post the diagram of your gate and we'll try find a solution

GPL71, 2019-01-08:
@GPL71 You must create an ambient and include the accessory in off state. Then, create an automation "When switch on -> exec ambient".
@RavenSystem Thank you ... I'll try as soon as possible ... thank you

Codi878, 2019-01-11:
> KellyTron and 0.5.7 are same code for Shelly1. Forget KellyTron.
> 
> Try RavenCore 0.5.8 and set external toggle to 0 for testing.

Hi Jose 

On 0.5.8 firmware shelly1 it does not work properly. I tested in various ways and unfortunately is more often unavailable in the home app on 230v

RavenSystem, 2019-01-11:
@Codi878 Next step will be to deactivate UART (with a custom option in EVE) and verify if that is the problem.

RavenSystem, 2019-01-14:
@Codi878 About Shelly 1: update to last version, please. And then set "Log output" to 0 and reboot your device.

Codi878, 2019-01-14:
Hi Jose,

Now Shelly1 is not visible on home app and I can't add new accessory.

RavenSystem, 2019-01-14:
@Codi878 What? Has accessory disappear from Home App? Nothing in the code can do that.

Codi878, 2019-01-14:
I erase the module and uploaded the firmware again, and unfortunately, shelly does not appear to be paired with homekit. I tried several times and the same thing each time. Initially looks like some Bug maybe someone is able to confirm it?

Codi878, 2019-01-14:
When I upload Tasmota firmware shelly works fine.

RavenSystem, 2019-01-14:
@Codi878 If you erase firmware, HomeKit config is removed, and you must pair it again.

Try erasing it, flash it again, and config wifi, HomeKit and device type BEFORE connecting it to 230VAC.

Codi878, 2019-01-14:
That's what I did as you wrote, unfortunately it does not appear to be re-pairing

RavenSystem, 2019-01-14:
Can you post logs?

RavenSystem, 2019-01-14:
Are you using v0.5.10?

Codi878, 2019-01-14:
Will try it tomorrow, connect it and send logs
Thank You

RavenSystem, 2019-01-14:
@digiservice2001 Test it connected to main. I have one Shelly1 working in my house without any issue, but @Codi878 has many stability problems when he uses it with main.

@Codi878 Try version 0.5.11 that I have released just now.

For Shelly devices, it is recommended to set "Log output" to 0 in EVE settings to disable UART when connected to 230VAC. And of course, setting "Show setup" to OFF is always recommended to keep HomeKit accessories free of extra work.

Codi878, 2019-01-15:
Hi Jose 

This is a log from shelly1

Cache-Control: no-cache
Vary: X-PJAX
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/118357494/829d2480-1077-11e9-9664-ecd2e3d762ea?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190115%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190115T133505Z&X-Amz-Expires=300&X-Amz-Signature=3c9f4c8491c59ce542e6cdaceb748d79c758ff7ae6a77c09feeec6c7c074c264&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dotaboot.bin.sig&response-content-type=application%2Foctet-stream
Set-Cookie: has_recent_activity=1; path=/; expires=Tue, 15 Jan 2019 14:35:05 -0000
Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Sat, 15 Jan 2039 13:35:05 -0000; secure; HttpOnly
Set-Cookie: _gh_sess=cTNWNzJEendjKzRLcE8raEcwQWVTVUs4TzhCOTVic2U3eHVqdFd0eFExeW9TOGV5YURIamtQS3dZSVB6dnR
HTTP returns 302
--- ota_connect
target IP is 52.216.171.43 bind socket 55222....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github-production-release-asset-2e65be.s3.amazonaws.com port 443....OK
send request......OK

72 a2 94 95 ... 94 03 00 a7    
so far collected 154 bytes
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/ota/releases/download/0.3.0/otamain.bin.sig HTTP/1.1
Host: github.com

--- ota_connect
target IP is 140.82.118.4 bind socket 55223....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HTTP/1.1 404 Not Found
Date: Tue, 15 Jan 2019 13:35:12 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Server: GitHub.com
Status: 404 Not Found
Cache-Control: no-cache
Vary: X-PJAX
Set-Cookie: has_recent_activity=1; path=/; expires=Tue, 15 Jan 2019 14:35:12 -0000
Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Sat, 15 Jan 2039 13:35:12 -0000; secure; HttpOnly
Set-Cookie: _gh_sess=aWlwYUR2QTlrVGV4YjIrM1ROTVhOZGdkNnUrWWJxZWEyZ3lwdjhMWDV4Ykx0VExuN243LzU1cVhkMmtuNnliOEIrM2JBTEdnTUVFZTZsSko3bytLdnlkWGpwRUpuRkV3d2Zrc0NoQWx5S2RPNHNmZjh3OVViNEtKbVh1ZmhOdGVWem1ONHNQVWU0WU5OSFZCV0dPdFpXa3RKTmxYSktyWGFuZGNxVTFXank3WjlzL1lMTTE4K3FYY3pGNXdZVk5nd1kwSWU5Z0VFRSs2bDJ4aWZVbWZLdz09LS1PWSsvckk4OE5qZHppdVRoYVZxZ2pBPT0%3D--f6c7da88d45738b1cd53d6ba21169593d999eedc; path=/; secure; HttpOnly
X-Request-Id: 9b5630d1-46c8-4811-ba54-0c6b88094714
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection: 
HTTP returns 404
--- looping

RavenSystem, 2019-01-15:
@Codi878 This is a log from an old version of otaboot. Go to the Wifi and try again the new otaboot.bin

Codi878, 2019-01-15:
Ok now I have this:

>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
mode : sta(84:f3:eb:32:c9:5e) + softAP(86:f3:eb:32:c9:5e)
add if0
Function called without core lock
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=LCM-32C95E
bcn 0
del if1
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DHCP server
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
pp.c 603
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: ac:1f:74:20:7b:37 join, AID = 1
station: ac:1f:74:20:7b:37 leave, AID = 1
rm match
Timer Start Failed
Timer Stop Failed
Timer Start Failed
Timer Stop Failed
Timer Start Failed

RavenSystem, 2019-01-15:
This is a buf from LCM (OTA system), not Ravencore. Repower it and try again.

Codi878, 2019-01-15:
>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=LCM-32C95E
bcn 0
del if1
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DHCP server
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
scandone
scandone
scandone
scandone
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
scandone
scandone

Codi878, 2019-01-15:
I can't connect shelly1 to wifi

Codi878, 2019-01-15:
>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=LCM-32C95E
bcn 0
del if1
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DHCP server
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
scandone
scandone
scandone
no buf for probe, ie len 0
no buf for probe, ie len 0
scandone
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
scandone
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
no buf for probe, ie len 0
no buf for probe, ie len 0
no buf for probe, ie len 0
scandone
add 1
aid 1
station: f0:18:98:08:e0:13 join, AID = 1
station: f0:18:98:08:e0:13 leave, AID = 1
rm match
no buf for probe, ie len 0
scandone

RavenSystem, 2019-01-16:
@Codi878 Your issue is related to this: https://github.com/RavenSystem/esp-homekit-devices/issues/187

closed 2019-01-27 by RavenSystem