RavenSystem/esp-homekit-devices #173: Feature Request: Action push light

phlexx, 2019-01-08:
Hi!
Would be possible to add a new mode for using with a pushbutton circuit?  Eg. Common Hallway Lights in Europe with a Timer. This would of course require the dry switch mod.

I have the exact setup for testing purposes running as a garage door. 
Basically i set the opening duration as long as the light is on and "opening/closing the door" activates the light.

Main Features can be adopted:
- HMServiceType Lightbulb
- Main toggle triggers impulse (similar to garage door) and turns the light/fan/pp/switch on until timer runs out 

Settings:
- Slider for setting the duration before the light turns off (similar to garage door)

It should be very easy to build and would be an useful feature for me and hopefully many others. :)

Thanks for all your awesome work!

peros550, 2019-01-08:
+1

The slider to control the timer duration is a nice idea which would widely extend the uses cases. 
Of course an automation could be build to achieve the same result but I believe it's better to off load these simple automation from Apple TV.

phlexx, 2019-01-08:
@peros550

Thanks. Although I meant for the timer just to be a dummy “light is on”-state so you know if it is on and maybe the switch won’t allow a new trigger.

GPL71, 2019-01-08:
@phlexx
I'd wrote something about here
https://github.com/RavenSystem/esp-homekit-devices/issues/166

In italy we've lights in common hallway under an hardware timer, then a SONOFF SV with dry contacts is enough to connect at main equipment

And I wrote garage door device is similar the function required

phlexx, 2019-01-08:
After reading #166 a few things to clarify:
I just want to automate an impulse relay which is connected to a hardware timer. 

- The duration setting should be set accordingly to the hardware-timer so the state of the light is reflected correctly. So the bulb icon is on as long as the light is. The switch may or may not accept a new trigger. 

- The duration of the impulse(=turn on) can be set but is optional (keeping it at 1 second).

It could also be done with a simple impulse-switch which turns of after the impuls was set and no settings required.

This should work exactly like the garage-door mode. 

@GPL71 
This hallway-light situation is exactly what i want to do. But i dont want to setup the rule in homekit (turn off when turned on) because its to unstable to mess with the light of the whole building.
Just trigger it.

dvelardi, 2019-01-08:
> Risposta: RavenSystem/esp-homekit-devices <reply@reply.github.com>
> Data: martedì 8 gennaio 2019 12:36

Yes, also for me it’s better to be able to set time duration via firmware, so it’s independent from homekit /internet connection failure…

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

peros550, 2019-01-08:
@phlexx 

I've read your explanation. If I understand correct, you would like the icon's status to be completely separated to the actual relay output , correct?

You would like the icon in the home app to stay on as long as the hardware timer is programmed to stay on (configurable through eve app) 

and you would want the relay output to work like the garage door example   

I share with you the same view on automation's experience. Apple TV is good, but if it can be programmed within the chip, it's way better and effortless.

phlexx, 2019-01-08:
@peros550 

100% correct. :)
But im pointing out that the icon-virtual-operation-thing is purely a nice-to-have thing for the timer!

Optionally you **could** incorporate a slider for the the duration of the impulse.
This is just for other creative usecases some people might have. I personally don't need it.

homkittoby, 2019-01-10:
we need a new software option. with a rule - if on then off, does not work reliably and fast enough!

Idea for push button.
Switch will turn off automatically after invoke Timeout.

Setup Option
Push Button: on/off
Time for turn off: 0,5 - 10sec

See GPIO-PushButton-Service @ https://www.npmjs.com/package/homebridge-gpio-services

https://github.com/spobel/homebridge-gpio-services/blob/master/lib/accessories/pushbutton.js

See issuse Relay Trigger Question #47

phlexx, 2019-01-10:
@homkittoby thanks for your input.
Looks like i stumbled a upon an ongoing request.

After reading #47 i understand the principle of @RavenSystem of only implementing native homekit features and i respect that.

But i wanna point out that this feature not necessarily goes against the rules of homekit.

The people want a switch which only fires current for <duration> and the setting to set <duration>.
This can be a simple switch. No Push button device. Just a switch you can turn on und it turns itself of after <duration>.

Please, Jose would you be so kind and give as a short statement?
(Im feeling weird when i have to tell siri to open the garage door just so i can put the keys in the lock.)
Thanks very much in advance

RavenSystem, 2019-01-10:
I see a lot of people want this. Ok, this feature will be supported with v0.6.X and ONLY for device types using one relay as switch.

- Min value will be: 0, meaning disabled.
- Step value will be: 0.2 seconds.

Please, discuss here what must be MAX value.

homkittoby, 2019-01-10:
@ravencore

Thank you 🙏🏻

0,2-5sec are very nice!

Idea for the new function.

Version for
3. Socket + Button and
9. Socket + Button + TH sensor and
10. ESP01 Switch + Button.

Relay with auto off after 0.2-5sec &
Button Mode only, without function to switch the relay!

You are doing a great job 🤠

ymyshko, 2019-01-11:
Min value 0 - meaning disable
Max value - 180s
Step 0.5s is enough. But 0.2s is good too.
Short time can be used to control electric locks.
Long time can be used to control light in transit rooms as a hardware timer.
Sonoff basic original firmware has inching mode from 0.5 to 3600s

peros550, 2019-01-11:
Max value of 3600 seconds could also be of use to me. 

I have a sonoff activating a power relay which in turn activates the water boiler. Although there is a hardware thermostat , I forget some times to turn  off the Sonoff. 

I have solved this problem , but I use a raspberry , HomeBridge , virtual switches and a couple of automation rules in the Apple TV . 

Having a hardware auto off timer implemented in the Ravencore would simplify and make it more efficient

GPL71, 2019-01-24:
@RavenSystem
Hello.... I'm forwarding 1 link to youtube with sample about inching (temporary close of a button) and switch (wall mount switch)

In the case below it is used a board with double hardware function (temporary switch / stable switch)
https://www.youtube.com/watch?v=i_Le7qZ5iBI

I've seen another video but I can't found at moment..
Using EVE you could select the function temporary button or bi-stable switch (ON -OFF) live without reboot.
And Relè stay on for the time you tap the icon

Thank you for marvellous great work

RavenSystem, 2019-04-13:
Supported in RavenCore v0.8.2.