RavenSystem/esp-homekit-devices #174: 0.5.8 toggle fail?

Manelvallsvila, 2019-01-08:
Hi Jose,

Thanks in advance for the good job you are doing!

Since last Update to 0.5.8, on a sonoff basic, the mode 2 toggle doesn't work with the Wall switch. I have a classic sonoff board with Wall switch conected to GND and gpio14.

Question:
Is it posible to be a software problema? I have this unit in Wall...

Another question:
Every time you reléase a new firmware versión , I flash it OTA, but I don't know if it is necessary or if it's not.
How can I know what changes you implement in new versions to know if its needed to flash or an option?

Muchas gracias!

RavenSystem, 2019-01-11:
Can you post your custom settings from EVE?

What you mean when say "I flash it OTA"??

FYI: Changelogs are here: https://github.com/RavenSystem/esp-homekit-devices/releases

macjl, 2019-01-11:
Same issue here. With this settings, a button between GPIO2 and GND doesn't change the switch state on a Sonoff Basic: 
- Board Type : 2
- Device Type : 1
- External Toggle : 1
- Init State SW1 : 0

Manelvallsvila, 2019-01-12:
My custom settings feom EVE are:

Board Type: 1
Device type: 1
extermal toggle 1: 2
Init state SW1: 0

RavenSystem, 2019-01-12:
I will test it asap.

RavenSystem, 2019-01-12:
@macjl there is a bug with button mode (external toggle 1 = 1). It will be fixed with next update. 

@Manelvallsvila I need some testing to know where is the issue with extertal toggle 1 = 2, because it must work.

RavenSystem, 2019-01-12:
@Manelvallsvila Sorry, I can not reproduce your issue.

RavenSystem, 2019-01-13:
@seritos What you mean exactly? When external switch is ON (using External toggle: 2), Sonoff doesn't boot?

seritos, 2019-01-13:
If any of these two switches is pressed before the sonoff has power, then the sonoff wont boot/start once you power it up.
I discover it yesterday. I was doing some test and had to disable the mains from home to operate safely. Once finished, realized that some of the Sonoff where unresponsive. 
Then I relized why.

RavenSystem, 2019-01-13:
@seritos Confirmed! If GPIO2 is low (connected to GND), device doesn't boot anything. It's a hardware "issue" and can not be solved with code.

For an ESP module to boot up into normal run mode then the pins must be:
- GPIO0 = high
- GPIO2 = high
- GPIO15 = low

https://www.esp8266.com/viewtopic.php?t=18550

RavenSystem, 2019-01-13:
@seritos I think the best way to fix is replacing your Sonoff Basic with an old version, or a Shelly1.

seritos, 2019-01-13:
Oh, what a pity. 

I will try to solder it directly to the chip then.

Thanx!

RavenSystem, 2019-01-27:
@seritos Version 0.6.0 has a new board type 4, and you can attach your external toggle to GPIO3/RX pin.

closed 2019-01-27 by RavenSystem