RavenSystem/esp-homekit-devices #177: ESP8266（nodemcu）

dydingjiawei, 2019-01-10:
Can I flush RavenCore to ESP8266 (NODEMCU)?I can connect to the homekit,but I can't find the controlled GPIO port, I want to control the relay.

dydingjiawei, 2019-01-10:
> If you set the device to "Device Type 10" then you have to connect the relay to GPIO2 which is labeled on the NodeMCU as D4. If you want a Button too, you have to connect it to GPIO0 -> D3.
> 
> For more information take a look at the Wiki under Device type 10

Emmmm, how to change the Device Type??😂😂

yukonx, 2019-01-10:
LED_GPIO                        13
DOOR_OPENED_GPIO                4
DOOR_OBSTRUCTION_GPIO           5
DOOR_CLOSED_GPIO                14
TOGGLE_GPIO                     14
BUTTON1_GPIO                    0
BUTTON2_GPIO                    9
BUTTON3_GPIO                    10
BUTTON4_GPIO                    14
RELAY1_GPIO                     12
RELAY2_GPIO                     5
RELAY3_GPIO                     4
RELAY4_GPIO                     15

**change the Device Type**

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#user-settings

dydingjiawei, 2019-01-10:
[image]
In afct, i can't control my realy, The Relay is benn connected to GPIO0 (D3)
[image]

yukonx, 2019-01-10:
> [image]
> 在afct，我無法控制我的真實，繼電器已連接到GPIO0（D3）
> [image]

type1&10   RY use gpio2

dydingjiawei, 2019-01-10:
[image]
[image]
I have tried it and still not working.

dydingjiawei, 2019-01-10:
[image]
When I press the FLASH button on the ESP8266, my phone will change.

yukonx, 2019-01-10:
> [image]
> When I press the FLASH button on the ESP8266, my phone will change.

我也是使用 ESP8266 去做的 你的GPIO 因為程式的限制要改成 gpio12 ,gpio0 是用來閃存或當作按鈕用的

yukonx, 2019-01-10:
> > 我也是使用 ESP8266 去做的 你的GPIO 因為程式的限制要改成 gpio12 ,gpio0 是用來閃存或當作按鈕用的
> 
> 所以我要连接到GPIO12吗？还是不起作用，我是刷的main.bin,blank_config.bin,rboot.bin

程序是沒問題的 要用gpio12 才會動作  你用GND+gpio0 看看iphone上狀態會不會改變,會就是程序正常

dydingjiawei, 2019-01-10:
> > > 我也是使用 ESP8266 去做的 你的GPIO 因為程式的限制要改成 gpio12 ,gpio0 是用來閃存或當作按鈕用的
> > 
> > 
> > 所以我要连接到GPIO12吗？还是不起作用，我是刷的main.bin,blank_config.bin,rboot.bin
> 
> 程序是沒問題的 要用gpio12 才會動作 你用GND+gpio0 看看iphone上狀態會不會改變,會就是程序正常

GND+GPIO0,看串口上的信息显示是板子直接重启服务了

yukonx, 2019-01-10:
> > > > 我也是使用 ESP8266 去做的 你的GPIO 因為程式的限制要改成 gpio12 ,gpio0 是用來閃存或當作按鈕用的
> > > 
> > > 
> > > 所以我要连接到GPIO12吗？还是不起作用，我是刷的main.bin,blank_config.bin,rboot.bin
> > 
> > 
> > 程序是沒問題的 要用gpio12 才會動作 你用GND+gpio0 看看iphone上狀態會不會改變,會就是程序正常
> 
> GND+GPIO0,看串口上的信息显示是板子直接重启服务了
> 然后iPhone上homekit显示该配件无反应，我必须要重新刷一边固件才可以添加上去，这个现象正常吗？

不正常,,我是直接用ESP12E做的 操作上都正常

dydingjiawei, 2019-01-10:
> > > > > 我也是使用 ESP8266 去做的 你的GPIO 因為程式的限制要改成 gpio12 ,gpio0 是用來閃存或當作按鈕用的
> > > > 
> > > > 
> > > > 所以我要连接到GPIO12吗？还是不起作用，我是刷的main.bin,blank_config.bin,rboot.bin
> > > 
> > > 
> > > 程序是沒問題的 要用gpio12 才會動作 你用GND+gpio0 看看iphone上狀態會不會改變,會就是程序正常
> > 
> > 
> > GND+GPIO0,看串口上的信息显示是板子直接重启服务了
> > 然后iPhone上homekit显示该配件无反应，我必须要重新刷一边固件才可以添加上去，这个现象正常吗？
> 
> 不正常,,我是直接用ESP12E做的 操作上都正常

我也不清楚我板子上的WIFI模组具体型号，没有标注。

yukonx, 2019-01-10:
> > > GND+GPIO0,看串口上的信息显示是板子直接重启服务了
> > > 然后iPhone上homekit显示该配件无反应，我必须要重新刷一边固件才可以添加上去，这个现象正常吗？
> > 
> > 
> > 不正常,,我是直接用ESP12E做的 操作上都正常
> 
> 我也不清楚我板子上的WIFI模组具体型号，没有标注。

那個應該沒差    D3=GPIO0  D6=GPIO12 你有這樣接嗎?

dydingjiawei, 2019-01-10:
> > > > GND+GPIO0,看串口上的信息显示是板子直接重启服务了
> > > > 然后iPhone上homekit显示该配件无反应，我必须要重新刷一边固件才可以添加上去，这个现象正常吗？
> > > 
> > > 
> > > 不正常,,我是直接用ESP12E做的 操作上都正常
> > 
> > 
> > 我也不清楚我板子上的WIFI模组具体型号，没有标注。
> 
> 那個應該沒差 D3=GPIO0 D6=GPIO12 你有這樣接嗎?

将两根阵脚直接连接起来?

yukonx, 2019-01-10:
> > 那個應該沒差 D3=GPIO0 D6=GPIO12 你有這樣接嗎?
> 
> 将两根阵脚直接连接起来?我试试

不用我只是確定一下你有沒有接對,等我下班回家再照我的給你參考

dydingjiawei, 2019-01-10:
> > > 那個應該沒差 D3=GPIO0 D6=GPIO12 你有這樣接嗎?
> > 
> > 
> > 将两根阵脚直接连接起来?我试试
> 
> 不用我只是確定一下你有沒有接對,等我下班回家再照我的給你參考

好的，那就麻烦大佬了

yukonx, 2019-01-10:
> 好的，那就麻烦大佬了

我接這樣就可以用了，你用那個板子應該也是一樣就可以了

https://drive.google.com/open?id=1T945lJFUvOM2cngdB_uuS1zvU8Qll2n8
https://drive.google.com/open?id=1a-AqF2GiT4QJ3imrbnmUw42Sg9Kg7a9u
https://drive.google.com/open?id=1aSLMF20UxHmpVccRMO0UoVJqOglo6F6U

影片

https://youtu.be/Ly2AtoLystg

dydingjiawei, 2019-01-10:
> > 好的，那就麻烦大佬了
> 
> 我接這樣就可以用了，你用那個板子應該也是一樣就可以了
> 
> https://drive.google.com/open?id=1T945lJFUvOM2cngdB_uuS1zvU8Qll2n8
> https://drive.google.com/open?id=1a-AqF2GiT4QJ3imrbnmUw42Sg9Kg7a9u
> https://drive.google.com/open?id=1aSLMF20UxHmpVccRMO0UoVJqOglo6F6U
> 
> 影片
> 
> https://youtu.be/Ly2AtoLystg

我才发现一个问题，我的继电器好像突然坏掉了，前天还好好的。

closed 2019-01-27 by RavenSystem