RavenSystem/esp-homekit-devices #178: Sonoff TH Reset Not Working

RickNunn, 2019-01-10:
I have a Sonoff TH working as I want, but I set it up and tested it in one "home" and now I want to move it to another, on another wifi. 

I have attempted to follow the reset procedure of holding the button down for 10 seconds, but isn't working for me. The device stays on my current network, and doesn't start creating it's own.

See video here — https://www.dropbox.com/s/83cj85x6bbyn57f/Sonoff%20TH%20-%20No%20Reset.mp4?dl=0

Am I doing it wrong?

RavenSystem, 2019-01-10:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#resetting-device-to-factory-defaults

RavenSystem, 2019-01-10:
In the video device seems to reset fine to factory defaults.

RickNunn, 2019-01-10:
Thanks for the quick reply. Shouldn’t it have rest the WiFi settings? 

That’s my aim.

RickNunn, 2019-01-10:
Nope, that’s not showing up for me

RickNunn, 2019-01-10:
Potentially the flashing it does in the video is because i didn’t have the sensor connected?

RavenSystem, 2019-01-10:
Can you post here your RavenCore custom settings? (What EVE shows in setup)

RickNunn, 2019-01-10:
[image]

RavenSystem, 2019-01-10:
Are you using v0.5.8?

RickNunn, 2019-01-10:
I sure am

RavenSystem, 2019-01-10:
It must work. Try this:
- connect dev to ac power. 
- wait 30 seconds
- push and hold button for at least 15 seconds. 
- record all in video.

RickNunn, 2019-01-10:
Hey that worked, excellent. Thank you so much, those extra seconds must have made all the difference. It's now successfully added to my Homekit on a different wifi network. 

Do you have a tip jar or anything setup for this project?

closed 2019-01-11 by RickNunn

RavenSystem, 2019-01-11:
I have an Amazon WhishList: https://www.amazon.es/gp/registry/wishlist/1S1G3PBZ328ZD/

RickNunn, 2019-01-15:
I tried to use the Wishlist but, I think, because they are being sold by 3rd parties it wont work without third-party address sharing being on.

From Amazon:

> If you're purchasing an item from a seller, the default delivery address will only be available if the Wish List owner has enabled third-party address sharing. Otherwise, you'll need to know and enter the delivery address.

Totally understandable if you don't want to turn that on. But let me know if you do.

Thanks

RavenSystem, 2019-01-15:
@RickNunn I enable it.