RavenSystem/esp-homekit-devices #180: Sonoff Thermostat mode

Codi878, 2019-01-11:
Hi Jose

I use sonoff TH modified for dry contact to control the gas boiler and from my observations after setting the set temperature quite often the boiler turns on and off. 

Is it possible to implement such a solution as for example in this project:

https://www.npmjs.com/package/homebridge-pi-thermostat

"minimumOnOffTime": 120000,
 "blowerTurnOffTime": 80000,
 "startDelay": 10000,
 "temperatureCheckInterval": 10000

A protection against too frequent switching on was used here and it would be nice if you could configure Sonoff TH in RavenCore firmware in the hidden Eve menu

Codi878, 2019-01-22:
Hi Jose,

Could you add this upgrade to your road map? 

is there an option to add such a solution to thermostat?

RavenSystem, 2019-01-22:
Don't worry. In the new firmware there will be Thermostat (2 relays), AC (1 relay) and Heater (1 relay) accessory types.

Kristian8606, 2019-01-22:
In my example, I added a pause of 0.2 degrees

closed 2019-01-27 by RavenSystem