RavenSystem/esp-homekit-devices #184: Installation problem:(

jascdk, 2019-01-13:
Got problem - not the programming novice:)

I tried to install the RavenCore on a Wemos D1 mini. It worked well, but I can't figure out what I have to do with the OTA step.

What should I fill in the last boxes in the setup screen with wifisetup?

How do I edit the files, so it suits my setup?

All the best

Jacob

jascdk, 2019-01-14:
Yes I read that, but when going through the EVE app, i cant see how I make - for example an Sprinkler Accessory. Is this firmware only for the sonoffs or? The test i made is by using a wemos D1 mini - it first showed up in the Apple Home App. There i had the options for making a light (bulb), a fan and a switch. Then i went over to the EVE app, but that did´nt help me. DO you know, if there is any videos that explain how to set it up?

digiservice2001, 2019-01-14:
HTTP/1.1 404 Not Found
Date: Mon, 14 Jan 2019 19:08:10 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Server: GitHub.com
Status: 404 Not Found
Cache-Control: no-cache
Vary: X-PJAX
Set-Cookie: has_recent_activity=1; path=/; expires=Mon, 14 Jan 2019 20:08:10 -0000
Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Fri, 14 Jan 2039 19:08:10 -0000; secure; HttpOnly
Set-Cookie: _gh_sess=eUFKTk5kQVhNQVhzVmlhWVVkRjNiRTAyektZQnR3YWc2VFAzZ1pWWHNJUVRPbS9HTFZRWXg4blJWVGpDVEFsUU9iZHkyNVlvMzc0ZnZ0NndUZWQrWXVRaXZoUnAyazFDbzI4aldNWUdoOGFWeG5RanZ2WHQ3WE1XSzZYTTVibmtCTFFQYmVreEl6NWhWUFZFZmd2cGpzcTR1MTE3SUQwZVBCYmV5aEdrYm9iZXlIY1F1aDJ3NG55dzkxL0g3a2h4dENtQmRjWVpmMnVaSSsvbmlkK0lsZz09LS1RZzVxL25tOGdpRS9JZ09waXdJb29BPT0%3D--46ab24ca169bda4d6841c343865da5aaa388150e; path=/; secure; HttpOnly
X-Request-Id: fe501884-c51f-45ad-a55c-5d1876b2ceff
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection: 
HTTP returns 404
--- looping

digiservice2001, 2019-01-14:
José, could you solve the missing part issue, please?
Thanx!

RavenSystem, 2019-01-14:
@digiservice2001 It is a LCM bug and I don't know where is the problem. You can open an issue into LCM repo: https://github.com/HomeACcessoryKid/life-cycle-manager

Or try to update, install it again.

digiservice2001, 2019-01-14:
I was lamer... Does not overwrite otaboot.bin.... :-))

gerarddvb, 2019-01-17:
Not related to the issue.. But can someone explain how to get this data?
I always wait 10 minutes and if it doesn’t show in HomeKit I reflash it and start over. Would love to see some output during during ota install

RavenSystem, 2019-01-17:
@gerarddvb Open a Terminal and exec: `nc -kulnw0 45678`

closed 2019-01-27 by RavenSystem