RavenSystem/esp-homekit-devices #187: 0.5.11 can't flash

St0nehead, 2019-01-15:
Tried to erase my Sonoff Basic and reflash to new 0.5.11 firmware.
No success at all!
After 1st reboot i connected to WiFi LCM-B507XXX... DHCP doesn't work.
IP shown in prefernces is 169.254.74.23 (255.255.0.0) -- self-assigned
No captive portal! 
$: screen /dev/tty.SLAB_USBtoUART 115200 shows following info:

scandone
scandone
scandone
station: 4c:32:xx:xx:xx:xx leave, AID = 1
rm match
scandone
add 1
aid 1
station: 4c:32:xx:xx:xx:xx join, AID = 1
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
station: 4c:32:xx:xx:xx:xx leave, AID = 1
rm match
scandone
add 1
aid 1

;'(((((((

digiservice2001, 2019-01-15:
Restart sonoff after flash, before connecting to LCM-xxxxx AP may help.

St0nehead, 2019-01-16:
#digiservice2001 Thank you! I restarted my sonoff after flashing and now all seems to be ok!

#GPL71 Yes, i’m a mac user

St0nehead, 2019-01-16:
#offtop How to reconfigure WiFi settings, after toogle “Reset WiFi” switch in Setup menu?