RavenSystem/esp-homekit-devices #189: Window Blinds Accesory

seritos, 2019-01-15:
Hi @RavenSystem. I know you will release a version that supports the blinds.
Would be possible, for now, to simply create the accesory? Not the securtity control to avoid the double selector. 
I have implemented a relay to avoid this situation, so the basic accesory will be enough.
Is this possible? 

Thanx!

GPL71, 2019-01-16:
to avoid relay ... you can create an ambient you'll call up and another called down.
Each one of them will manage the swicth off of other channel

GPL71, 2019-01-16:
sorry ...
when you activate an ambient the first channel turn on and the second turn down.
If you choose use 2 BASIC for the moment you can choose garage door option and do the ambient UP and DOWN

mgoeller, 2019-01-16:
I have prepared some Code for a Shelly 2 roller shutter, see attachment. In case this is of any help tp you, @RavenSystem. I am also open for improvement hints :)
[main.txt](https://github.com/RavenSystem/esp-homekit-devices/files/2763785/main.txt)