RavenSystem/esp-homekit-devices #190: Problem with Shelly 1 Switch

raffi961, 2019-01-16:
Hi

I have flashed the firmware of my shelly 1 with the latest firmware and added it to Homekit and all seems to be working right except the external toggle button

My setting in EVE are the following

Board type : 3
Device Type: 1
External toggle 1: 2
init state : 0
log output : 0

turning on and off from the phone app works great, but the actual swithc on my wall doesn't do anything

What's wrong ?

RavenSystem, 2019-01-16:
It is a bug. I am recoding somethings to reduce flash size and these things can occur.

RavenSystem, 2019-01-16:
Please, update to version 0.5.15 and comment.

raffi961, 2019-01-16:
Just reflashed it and this time before adding it to homekit , i disconnected/reconnected and it worked. now testing the switch

raffi961, 2019-01-16:
There is a Bug, the switch started working but here is what happens. tested the scenario 2-3 times and the same happens.
Turn on from app -> turns on
Turn Off by the switch -> turns off and updates the icon to off in home app
Turn back on from the switch -> turn on and updates the icon to on in home app BUT an orange exclamation mark appears in the corner and stays there and i can no more turn off or on from the app

BUT if i turn on off from the switch it actually updates the icon in the home app to show the status of the light.
But then eventually I get a NO RESPONSE

RavenSystem, 2019-01-25:
> @raffi961 @RavenSystem
> Hi boys ..... I have a question about bug dicovered.
> Is it solved now?
> just arrived 4 shelly and I want start test them
> Thanks
> Ciao

My Shelly1 works perfectly. Remember to complete setup it when connected to 3.3V.

raffi961, 2019-01-25:
Yes

GPL71, 2019-01-26:
@RavenSystem @raffi961 
just installed 4 shelly.
They're working great!!!! They're always connected to the net.
today I'd order 4 shelly 1 again
Question... Any issue with shelly 2?
I needed them too
Thank you
Ciao

GPL71, 2019-01-28:
@RavenSystem
Yes it's on back board unfortunatly. However it is possible.
If shelly doesn't show issue, I would like add this experience, because I'm afraid that using only more shally 1 may be I'll have problem with pairing.

RavenSystem, 2019-01-28:
@GPL71 You must not have any problem with pairing using more than one Shelly1. Each device has its MAC address, accessory ID and accessory name. You simply connect and configure one by one.

GPL71, 2019-01-30:
I'm planning to use about 20 Shelly1 in my new house. @RavenCore
wonderful ....

Ok then I'll purchase only shelly 1 ... the number is about 20 for me too

I'm looking for shelly 4 pro... it's a very interesting device.
 If I purchase one, what do you think ... Is it possible develope a homekit firmware?

Is there a way to be in touch with you private? I need speak with you ...
Thank you

GPL71, 2019-01-30:
I'm planning to use about 20 Shelly1 in my new house...... 
@RavenSystem for me too ... about 20 ... I'll plan purchasing of Shelly 1

I'm looking for 4 ch for my new house.... What do you think ... Is it possibile develope th firmware?
If so I would like purchase and offer me for testing.

Another request ... Is it possible be in touch with in private. I need speak with you.... thank you

RavenSystem, 2019-01-30:
@GPL71 email?

GPL71, 2019-01-30:
@RavenSystem econti.online@gmail.com

GPL71, 2019-01-30:
@RavenSystem email is ok form me and I wrote it, but I think system deleted it
How do I do ?
Thank you

RavenSystem, 2019-02-05:
@GPL71 Actually, the only thing that needs to be configured with Shelly 1-2 connected to 3.3V is "board type 3". If "board type" is not 3, or "device type" is not 1 or 2 (or 4 when Shelly4 is supported), Shelly will not boot.

Connected to 230VAC, you can reset wifi settings from setup without any problem.

peros550, 2019-02-05:
I'm very interested in these Shelly devices. 
If I get it right, once Shelly 1 or 2 is installed into a wall box, it should never change type right? Otherwise there will be a need for de-installing it out of the wall box and reconfigure it using serial? 

What if reset configuration happens? Will it keep device type setting?

RavenSystem, 2019-02-06:
@peros550 "Board type" is the unique value you must not change when connected to 230VAC. You can change external switch setup, firmware update, wifi reset, etc...

"Board type" is kept when a factory reset is done to avoid mess.

GPL71, 2019-02-07:
@RavenSystem
Just ended installation of the last 4 shelly1 delivered and flashed with 0.6.1
All working great.
I'm waiting for shipping of 4 shelly1 again .....

closed 2019-02-28 by RavenSystem