RavenSystem/esp-homekit-devices #192: Using GPIO3 with external standard wall switch (new layout)

georgia088, 2019-01-18:
In order to use a standard wall switch the way the old layout used GPIO14 to Gnd, I am trying to understand how to do it.   I am new to all this, so please bear with me. You are soldering wires from GPIO3 (RX) and GND to a standard Wall Switch (not momentary/rocker). Then setting GPIO3 to (GPIO3 Serial In - Switch1 (09)) in Tasmota?

Is this correct? Does it work as it should?

Thanks!!

RavenSystem, 2019-01-18:
I am considering remove board type 2, and let user to select extra GPIO number manually. GPIO2 is accessible, but has problems with boot if it is low. Of course, using GPIO3 will disable UART.

seritos, 2019-01-18:
This option sounds just perfect.
Will be able to use more than one GPIO? 

Thanx!